S7NetPlus/s7netplus #171: Read StringEx fails if DB contains rubish

bauerju, 2018-07-30:
Today I had an exception while reading a String from the PLC. Someone wrote to the wrong DB. This destroyed the used string length information.   
Example: Tag was declared as DB1.DBB0 with DataType StringEx and Length of 4 chars. This takes 6 Bytes in plc. Something like that: 04 04 31 32 33 34. 

Someone wrote to DB1.DBB1 a value higher than  4. 
In the DB was something like that: 04 05 31 32 33 34. 

The read came back with 6 bytes. But the conversion to String failed in StringEx.From,ByteArray(Byte[] bytes).
Yes I know. This is something which should not happen.  
I changed the source for me this way:

        public static string FromByteArray(byte[] bytes)
        {
            if (bytes.Length < 2) return "";

            int size = bytes[0];
            int length = bytes[1];

            // Check if byte array is long enough
            if (bytes.Length - 2 < length)
            {
                length = bytes.Length - 2;
            }

            return System.Text.Encoding.ASCII.GetString(bytes, 2, length);
        }

Jürgen

mycroes, 2018-07-31:
Hi @bauerju,

Honesty I think an exception would be in place here, since the data is invalid. Masking that by returning the garbage as string or returning an empty string isn't always a good idea, although I can image we could create a `bool TryConvertFromByteArray(byte[] input, out string value)` on `StringEx` to allow for the behavior you implemented (although I would return `null` when `length < 2`). I see that the current exception has the following message: **Index and count must refer to a location within the buffer.**, wrapping this in another exception with a message that the string conversion failed would seem to address this issue if you ask me.

So in short, my proposal:
1. Wrap the exception with a concise message
2. If that doesn't float your boat, add `bool StringEx.TryConvertFromByteArray(byte[] bytes, out string value)`

mycroes, 2018-07-31:
To amend my previous message, I just noted that the current code returns `""` on `bytes.Length < 2`, so ignore that comment.

bauerju, 2018-08-01:
Hi @mycroes 
Yes, you are right. An exception should be fired. But only if this is a single read. Just one item.
In the case of a read with multiple items, my opinion is that the dataItem should better get an error flag not to break the whole read. 

Jürgen

mycroes, 2018-08-02:
Hi @bauerju,

I agree about not breaking the rest of the read, so what I'll implement is to set all the fields that can be set and to throw an `AggregateException` to combine the individual errors at the end of the read action.

bauerju, 2018-08-03:
Hi @mycroes 
What about this: Change the DataItem to hold an additional byte array. This byte array can only be written or read from plc. 
A read or write won't fail from used DataType.
Change the value property to a getter and setter which parses the object according to the used datatype from object to the byte array and back.  
You'll get the conversion error only if you try to access the value.  

Jürgen

scamille, 2020-08-25:
PR #292 is related to this

closed 2021-06-06 by mycroes