S7NetPlus/s7netplus #172: Trying to Open() needs a long time

padde77, 2018-07-31:
Hi there,

if a SPS is offline, Open() or IsAvailable() needs a long time (20 seconds).
Is there a way to try this fast? 

Thanks 
Patrick

bauerju, 2018-07-31:
Hi @padde77 
You could yourself send a ping with a timeout of 500ms to the ipadress of the plc.  
If the ping fails. There is no network communication with the plc. 
You can find Ping in namespace is `System.Net.NetworkInformation`. There are several examples in the net.
I use this myself. But I try it 3 times . If they all fail there is no communication.

Jürgen

padde77, 2018-07-31:
Hi @bauerju ,

thanks a lot, this is a good idea. I will try this, but until now I have this solution:

`var task = Task.Run(() => PLC1.IsAvailable);
                if (task.Wait(TimeSpan.FromSeconds(5)))
                    PLC1Online = true;
                else
                    PLC1Online = false;`

Thanks
Patrick

mycroes, 2018-07-31:
Hi @padde77 , @bauerju ,

It would also be possible to add a timeout to Open(), allowing fast(er) failure. Also, the `IsAvailable` check is a bad idea, a new connection is opened every time `IsAvailable` is consulted, which probably doesn't even work as expected since only a single connection can be openend with the connection ID used by S7NetPlus.

padde77, 2018-08-01:
Hi @mycroes ,
you are right, but I check my SPS before I run through my functions. That means, if IsAvailable() returns false, I do nothing, not even the Open(), because I do not need this after my IsAvailable().

mbalous, 2018-08-09:
Solution for this issue is implemented in PR #174 .

nanfengovo, 2025-02-20:
> 你好[@padde77](https://github.com/padde77)，[@bauerju](https://github.com/bauerju)，
> 
> 还可以为 Open() 添加超时，以允许更快（更快速）失败。此外，`IsAvailable`检查不是一个好主意，每次`IsAvailable`查询时都会打开一个新连接，这可能无法按预期工作，因为只能使用 S7NetPlus 使用的连接 ID 打开一个连接。

你好，请问怎么为Open()添加超时