S7NetPlus/s7netplus #176: Document is too old

NiHangBoy, 2018-08-14:
Since many changes have been applied. Please update you documents. Thanks a lot!

mycroes, 2018-08-14:
Hi @NiHangBoy,

This is a known problem. We're happy to accept contributions in this area. I'll see if I can devote some time to get the documentation in shape again on short term. Are there any specific areas you're missing in the current documentation that you'd like to see added?

FalcoGoodbody, 2020-09-09:
Hi @mycroes ,

I pick this "issue" because I am using the S7Net+ library very intensive and like it much.
My C#-skills are maybe something between beginner and advanced level and I do not feel that this is enough to support you and the others working on this library.
But maybe I can assist with updating the documentation/wiki? Would be finde do support in that way to give something back.
Do you mind that could be possible?

scamille, 2020-09-09:
Question is where do we want the documentation to go?

The docx/pdf seems a bit outdated and hard to keep in sync.
Maybe we want to use something like readthedocs to create documentation?
Or just put everything into the wiki.

FalcoGoodbody, 2020-09-09:
I would suggest to use the wiki, or? PDF is outdated, readthedocs is new to me but should not be a problem.

scamille, 2020-09-09:
I personally strongly prefer something with structured text like Markdown, like for example the wiki.

I haven't used ReadTheDocs myself as well, just thought it worth mentioning. 

https://dotnet.github.io/docfx/tutorial/docfx_getting_started.html is also a possibility if we want to focus on the API.

FalcoGoodbody, 2020-09-09:
Well I, as absolutely beginner here on Github, then should start with the wiki. It looks very simpel and easy to use.
Any hints what to do when starting editing the docu?

scamille, 2020-09-09:
Yeah: on the right, click on Add Custom Sidebar.

There create some sensible layout for the documentation. Maybe: Home/Introduction, Getting Started, Examples, etc.

Then for each section add a page to the wiki and write something :-)

https://github.com/simulationcraft/simc/wiki is an example which I know quite well.

mycroes, 2020-09-09:
Thanks guys, both of you have already been helping out a lot lately, it really feels like the project is getting back into shape thanks to you guys. You're both a big motivation to invest more time in the library and managing the project.