S7NetPlus/s7netplus #179: Cannot connect to Siemens Logo 0BA8

Zuendelmeister, 2018-08-21:
Hi all,

I cannot connect to my Logo 0BA8. As far as I know the logo uses CpuType S7200.
Here is the code:

`
Plc plc = new Plc(CpuType.S7200,ip, 0, 2);
plc.Open();
`
I also tried to vary rack and slot but it did not help.

Here is the exception:

`
S7.Net.PlcException: Couldn't establish the connection to <IP>.
Message: TPKT is incomplete / invalid ---> S7.Net.TPKTInvalidException: TPKT is incomplete / invalid
   at S7.Net.TPKT.Read(Stream stream) in C:\projects\s7netplus-wq17l\S7.Net\TPKT.cs:line 39
   at S7.Net.COTP.TPDU.Read(Stream stream) in C:\projects\s7netplus-wq17l\S7.Net\COTP.cs:line 53
   at S7.Net.COTP.TSDU.Read(Stream stream) in C:\projects\s7netplus-wq17l\S7.Net\COTP.cs:line 97
   at S7.Net.Plc.Open() in C:\projects\s7netplus-wq17l\S7.Net\PlcSynchronous.cs:line 32
   --- End of inner exception stack trace ---
   at S7.Net.Plc.Open() in C:\projects\s7netplus-wq17l\S7.Net\PlcSynchronous.cs:line 40
   at ConsoleApp1.Program.Main(String[] args) in <PATH>\ConsoleApp1\Program.cs:line 81
`

Ideas? Thanks in advance!

Best regards,
Zuendelmeister

mycroes, 2018-08-21:
Hi @Zuendelmeister,

Are you able to connect using other libraries? Unfortunately we will probably need some pointers to get this working. If you are able to provide a Wireshark dump of some working communication that should also get us somewhere.

Zuendelmeister, 2018-08-21:
Hi Mycroes,

thanks for your fast response! I can connect to the logo using node-red. I try to provide a wireshark log.

mycroes, 2018-08-21:
Hi @Zuendelmeister,

Could you try with [Sally7](/mycroes/Sally7) if you can connect with local TSAP 01:00 and remote TSAP 02:00? In a quick search those turned up as connection id's for Logo 8, while those are not being used for S7-200. We also don't support override of connection id's (see #153) at this moment, which could provide a workaround in S7NetPlus.

Zuendelmeister, 2018-08-21:
I cannot find a code example for Sally7 and I'm not familiar with TSAP and COTP. Sorry but I don't know how to use it.

mycroes, 2018-08-21:
I hope this will do (as example):
```csharp
var conn = new S7Connection("192.168.1.10", new Tsap(1, 0), new Tsap(1, 0));
conn.Open().GetAwaiter().GetResult();
```
Be sure to replace the IP address with the PLC address.

Zuendelmeister, 2018-08-21:
This did not work. I could create a connection with the ConnectionFactory though:
>
            var con = ConnectionFactory.GetConnection(ip, Sally7.Plc.CpuType.S7_200);
            con.Open();

Opening the connection worked as well. But reading values seems to be very complicated. For reading you need to create an object that implements the IDataItem interface. You even have to write the ReadValue function.

> struct dat : IDataItem
    {
        public Area Area => throw new NotImplementedException();
        public BigEndianShort DbNumber => throw new NotImplementedException();
        public Address Address => throw new NotImplementedException();
        public BigEndianShort ReadCount => throw new NotImplementedException();
        public TransportSize TransportSize => throw new NotImplementedException();
        public VariableType VariableType => throw new NotImplementedException();
        public void ReadValue(in ReadOnlySpan<byte> input)
        {
            throw new NotImplementedException();
        }
        public int WriteValue(in Span<byte> output)
        {
            throw new NotImplementedException();
        }
    }

But as I said the connection could be established.

mycroes, 2018-08-21:
@Zuendelmeister Thanks for testing, I was only concerned with the `Open()` call since that's the call that uses the local / remote TSAP. You can use `DataBlockDataItem<TValue>` to read / write data (without having to worry about conversion), but feel free to open issues for Sally7 at the respective [issues](/mycroes/Sally7/issues) page (I *am* the author of Sally7 and I *will* fix your issues if you report them there :wink:).

As far as S7NetPlus is concerned, I'll add initial support for Logo 8 by extending the `PlcType` enum, that should make it work.

Zuendelmeister, 2018-08-21:
Thats great, I'll test it out after you pushed the new version.
Thank you so much for your fast help!

mycroes, 2018-08-22:
@Zuendelmeister any chance you can fetch the code from #181, build it, and test if it works? There's two queue sizes that are negotiated and S7NetPlus was requesting size `3` for both, while Sally7 requests only `1`. Since concurrency is not supported anyway (not now at least) it's fine to set them to 1.

Zuendelmeister, 2018-08-23:
Hi @mycroes,

thanks for providing the new code. I tried it out but unfortunately the outcome is the same.
TPKT is incomplete / invalid

I tried again different racks and slots like so:
0,0
0,1
0,2
1,0
1,1
1,2
2,0
2,1
2,2

Do you have a clue what the problem could be?

mycroes, 2018-08-24:
@Zuendelmeister I did another comparison and found another difference. If you can fetch the latest code from the pull request you can test if it works. Besides that difference the messages from S7NetPlus and Sally7 are equal, so there's hope it works.

As far as rack and slot are considered, those are used for S7-300, 400, 1200 and 1500 only, not for S7-200.

Zuendelmeister, 2018-08-27:
I tried to check it out but I get an error:

git checkout -b develop 4c4dde8aeb046ffdf67cc7707a0320863e851b3b
fatal: reference is not a tree: 4c4dde8aeb046ffdf67cc7707a0320863e851b3b

So I copied the PLCHelpers.cs into my project from earlier, built it and ised the dll file. Unfortunately the outcome was still the same with the same error.

mycroes, 2018-08-27:
@Zuendelmeister You need to fetch the commit first:
```shell
git fetch origin 4c4dde8aeb046ffdf67cc7707a0320863e851b3b
git checkout -b [branch-name] FETCH_HEAD
```
Make sure to use the full commit hash for this, not the short hash. Also, `-b` is to create a new branch, `-B` allows for a reset of branch to the specified commit.

Anyway, I made some additional changes (still in #181). I was expecting the issue to come from the CommunicationSetup part, but perhaps it is originating in ConnectionRequest after all (just PLC that cuts off the connection a little later though). I checked the differences between Sally7 and S7NetPlus ConnectionRequest and there were some slight differences. I removed the differences in another two commits, I would like you to test both to see if it's perhaps fixed by removing the source reference only. Either way, it would be nice if this would fix the issue.

mycroes, 2018-08-27:
Added a third commit to fix the unit tests, you don't really need that for testing though.

Zuendelmeister, 2018-08-29:
I'll take a look as soon as I can.

mycroes, 2018-09-03:
HI @Zuendelmeister,

I found GitHub documentation that states another option (I'm not sure why you're getting the error though, but I can't help it):
```bash
git fetch origin pull/179/head:[branch-name]
git checkout [branch-name]
```
Could you give this a spin and see if that works? It should give you the latest commit in this PR.

Zuendelmeister, 2018-09-05:
I don't know why it does not work:
$  git fetch origin pull/179/head:develop
fatal: Couldn't find remote ref pull/179/head
(same for 181)
Can you provide a new branch with this commit? In this case I could checkout this one.

Zuendelmeister, 2018-09-17:
Anything new that I can try out?

mycroes, 2018-09-20:
@Zuendelmeister sorry for my absence on this issue. I was wrong with my fetch command regarding the **179**, that of course should have been **181**! It's strange that it doesn't work, but you can also download the commit as ZIP: [0d99f8fa...zip](https://github.com/S7NetPlus/s7netplus/archive/0d99f8fa2f4250cae2bda7511bed38a3aad4d814.zip). If you can confirm it works I'll merge that PR and do a new release straight away.

Zuendelmeister, 2018-09-24:
@mycroes 
thanks for providing the zip file. I cant get it to build the dll though.
The InitializeSourceControlInformationFromSourceControlManager seems to be missing.
[image]

mycroes, 2018-09-24:
What VS version are you using? The project is using the newer SDK style .csproj format, which I guess is only supported by newer VS 2017 versions.

Zuendelmeister, 2018-09-24:
I'm using a quite new Version
[image]

Zuendelmeister, 2018-09-25:
Can you maybe send me the dll for netstandard2.0? This way I can try it out.

Zuendelmeister, 2018-10-10:
Anything new?

mycroes, 2018-10-10:
Hi @Zuendelmeister,

Sorry for my late response. You should be able to download the [AppVeyor build of the PR](https://ci.appveyor.com/api/buildjobs/7ns15bkogy5fm387/artifacts/artifacts%2FS7netplus.0.4.0-b0011.nupkg), you can just extract it to get at the DLL's.

I'm really interested whether this works. Also, I'd be very happy to help you out with this, but not being able to fetch the changes could prove a bit troublesome. Doesn't mean it's an issue on your side, but I don't know a proper solution to that yet either (suggestions welcome).

Zuendelmeister, 2018-10-16:
Ok, I retried with the new dll but the outcome is the same.
Message: TPKT is incomplete / invalid ---> S7.Net.TPKTInvalidException: TPKT is incomplete / invalid

babinc, 2019-05-15:
Any update on this issue? I am having the same problem with a 300 series plc. Sally7 works.

mycroes, 2019-05-15:
Hi @babinc, is there any chance you could try the code from the develop branch? The issue was fixed in the end, but Sally7 actually still doesn't work for S7-200, because the major issue there seems to be the requested PDU size (which I should probably change in Sally7 as well...).

I didn't get to releasing the changes yet, but if develop does work for you (with S7-300) I'll do a release this week.

Regards,
Michael

noumanqaiser, 2019-07-16:
Hi Team, 
I was planning to use s7.net to read and write data to an Siemens logo plc, Can you confirm is S7.net can be used for Siemens Logo plc as of now?

Nouman

mycroes, 2019-07-16:
Hi @noumanqaiser, I still need to do that release, but then it should work. I'll try to do it this week, you could however test using the source as well if you don't want to wait for a release.

noumanqaiser, 2019-07-17:
Thanks MyCroes, No problem. I can wait for this week and hopefully I can help u with testing aswell on a logo PLC.

mycroes, 2019-07-18:
Hi @noumanqaiser, I released 0.4.0 yesterday, let me know if it works! If it doesn't work I want to fix it, but I can't test it myself obviously.

mycroes, 2019-07-22:
@noumanqaiser, did you get to testing it yet?

abduIkadir, 2019-11-04:
Hello,
I have same issue with ET200S IM151-8 PN/DP module and tried connection as logo8 and connected. 
I connected ET200 series modules as s7300 before. I wrote this here for information.

noumanqaiser, 2020-03-11:
HI @mycroes , Sorry for a really late reply. I finally got hold of a Logo OBA8 plc. I wanted to try out reading or wirting to this PLC but I am having a hard time finding out datablock number where I would read/write to. In Siemens Logo Sof Comfort software(software used to modify Logo Programs), it allows you to create basic function blocks and ladder diagram but it doesn't seem to show data block number/address where any data could be accessed/manipulated. 

Do you know how I could create a DB/Variable and obtain the DB number/offset value for a specific variable, that could really help.