S7NetPlus/s7netplus #182: Error Writing short.MinValue to INT DataBlock.

gpresland, 2018-08-30:
Writing a 16 bit INT (C# short)  to the PLC with a value of `short.MinValue`  results in the following error:

```
System.OverflowException: Value was either too large or too small for an Int16.
```

Writing `short.MaxValue` does not have any issue.

mycroes, 2018-08-30:
Hi @gpresland,

Thanks for reporting this issue. Could you provide the stack trace of the exception as well? `Types.Int` doesn't seem to have any issue dealing with `short.MinValue` (both to and from byte-array), so I'm curious what method you're using.

gpresland, 2018-08-30:
Hi @mycroes 

Looks like I made a mistake with the title; the issue is actually _reading_ the value `short.MinValue` from a `INT` DataBlock.

This is with a S1200.

You can reproduce the issue with the following:

```csharp
plc.Write("DB1.DBW258.0", short.MinValue);
var value = plc.Read("DB1.DBW258.0");
var castedValue = Convert.ChangeType(value, typeof(short));
```

Which results in:

```
An unhandled exception of type 'System.OverflowException' occurred in mscorlib.dll
Additional information: Value was either too large or too small for an Int16. occurred
```

This is occurring because the read returns `32768` instead of the expected `-32768` which is present on the PLC.

mycroes, 2018-08-31:
@gpresland The error is due to your `Convert.ChangeType` usage. The result of the `plc.Read` call is a `ushort`, with value `32768`. If you cast this to `short` it will work fine. Adjusted example:

```csharp
plc.Write("DB1.DBW258.0", short.MinValue);                  // Writes `-32768`
var value = plc.Read("DB1.DBW258.0");                       // Reads`32768`
var castedValue = (short) value;                            // Reads `-32768`
```

closed 2018-08-31 by mycroes

gpresland, 2018-08-31:
Changing the code to your example results in an equivalent error:

```
Additional information: Specified cast is not valid.
```

The value shouldn't be changing from a `short` to a `ushort`, as a `short` is present on the PLC. The value `-32768` is present on the PLC and reading that DataBlock shouldn't result in a different value.