S7NetPlus/s7netplus #183: Not an issue, I guess. Just a question.

Mallowan, 2018-09-06:
Hello,
It's getting to be ridiculous, however, I always get in trouble when I'm going to add a DB which I want to read.
For example, I have DB1 that is being read excellently. I added DB2 (did everything in the same way). Tried to read. I've got a strong sense, as though being read DB is not existing at all.
At the same time DB1 is being read normally.

 ```
public class DB1
    {

        public Double varReal1 { get; set; }
        public int varDInt1 { get; set; }
        public Double varReal2 { get; set; }
        public int varDInt2 { get; set; }
        public Double varReal3 { get; set; }
        public int varDInt3 { get; set; }
        public Double varReal4 { get; set; }
        public int varDInt4 { get; set; }
        public Double varReal5 { get; set; }
        public int varDInt5 { get; set; }
        public Double varReal6 { get; set; }

    }
```
 ```
public class DB2
    {
        public double varReal0 { get; set; }
        public double varReal1 { get; set; }
        public int varInt1 { get; set; }
        public bool varBool1 { get; set; }
        public bool varBool2 { get; set; }
        public bool varBool3 { get; set; }
        public bool varBool4 { get; set; }
        public bool varBool5 { get; set; }
        public bool varBool6 { get; set; }
        public bool varBool7 { get; set; }
        public bool varBool8 { get; set; }
        public bool varBool9 { get; set; }
        public bool varBool10 { get; set; }

        public bool varBool11 { get; set; }
    }
```
private void RefreshValues()
        {

            DB1 db1 = new DB1();
            plc.ReadClass(db1, 1); // reading DB1

            value0 = db1.varReal1;
            value1 = db1.varDInt1;
            value2 = db1.varReal2;
            value3 = db1.varDInt2;
            value4 = db1.varReal3;
            value5 = db1.varDInt3;
            value6 = db1.varReal4;
            value7 = db1.varDInt4;
            value8 = db1.varReal5;
            value9 = db1.varDInt5;
            value10 = db1.varReal6;

            DB2 db2 = new DB2();
            plc.ReadClass(db2, 2); // reading DB2

            db2_value11 = db2.varReal0;
            db2_value12 = db2.varReal1;
            db2_value13 = db2.varInt1;
            db2_value14 = db2.varBool1;
            db2_value15 = db2.varBool2;
            db2_value16 = db2.varBool3;
            db2_value17 = db2.varBool4;
            db2_value18 = db2.varBool5;
            db2_value19 = db2.varBool6;
            db2_value20 = db2.varBool7;
            db2_value21 = db2.varBool8;
            db2_value22 = db2.varBool9;
            db2_value23 = db2.varBool10;
            db2_value24 = db2.varBool10;

        }

mycroes, 2018-09-06:
Hi @Mallowan,

Perhaps you have **Optimize block access** turned on for DB2? I recommend reading a single variable at a known address, but honestly it shouldn't matter much since we're just reading bytes. If the type in C# doesn't match the type in the PLC, you'll just get garbage data, but it doesn't really change the ability to read or write values.

Mallowan, 2018-09-06:
Hi mycroes,
I studied "Optimize Block Access" feature, like nobody else. ) That's not a culptit!

Mallowan, 2018-09-06:
@**mycroes**, 
I did some It seem I've found where a library stumbled

mycroes, 2018-09-06:
HI @Mallowan 

> Now, I don't agree.
>
>I think something like a mismatch in types or so, completely halts reading of particular DB.
I did some study on that (now I would say an issue).
In other words, it seems I've found out where the library failed. For some reason the library doesn't like to read Integer itself or Integer in company with Reals. An usage of Int halts completely reading process, and there are not any exceptions there.
Compare both screenshots (above). If I change in DB2, in third line, an Int to DInt - a reading will go

I'm guessing the issue is with offset calculation. Probably if you extend your datablock at the end, you can read just as well (with incorrect values though!). There have been other reports about issues with offset calculation, but I've never been given a reproducible example. This might actually do though, so I'll see if I can find some time to test this.

Mallowan, 2018-09-07:
[image]

bauerju, 2018-09-07:
Hi @Mallowan 
Im my opnion your class should be changed:
instead of  
public int M_2 { get; set; }
you should use 
 public Int16 M_2 { get; set; } 
or 
 public short M_2 { get; set; }

Jürgen

Mallowan, 2018-09-07:
Juergen, 
I will try that. Hope it helps!

bauerju, 2018-09-07:
Hi @Mallowan 
No I'm not the developer of this library. Just a happy user. 
You need to use the exact type of variable corresponding to the used type in the DB.
S7netplus just reads bytes from the plc and maps it later to to the used variables.   
So if the types don't match, you'll get an different offset.

Jürgen

Mallowan, 2018-09-07:
> Hi @Mallowan
No I'm not the developer of this library. Just a happy user.
You need to use the exact type of variable corresponding to the used type in the DB.
S7netplus just reads bytes from the plc and maps it later to to the used variables.
So if the types don't match, you'll get an different offset.

Jürgen
> 

Yees, you are right. I did check-up on your advice. By default VStudio offers **Int** as Int(32) so that didn't  match with Step 7 Int that is to be 16 bit.

mycroes, 2018-09-07:
@Mallowan 
> I think you need to reread the title of my post more carefully.

I'm not sure why you keep thinking for others. I have read your issue title, that doesn't change a thing about my comment. And with regards to reading titles, *you* are posting this in the issue tracker for S7NetPlus. I honestly think you are being disrespectful (by actions, and sometimes slightly by what you type because you don't consider the position of the person you're addressing) to this project. If you keep that up I might act on that.