S7NetPlus/s7netplus #186: `IsAvailable` breaks PLC connection

mycroes, 2018-09-07:
As reported in #185, Using `IsAvailable` causes the connection to no longer work, thus leading to false errors in the next call that uses the connection.

andnig, 2020-01-31:
Hey guys, any updates on possible fixes for this issue?
Thanks :+1:

MianArrivas, 2020-07-06:
Hi mycroes!
Thanks to your library I am learning a lot more of PLC :D
Today I discovered about the problem of using IsAvailable property. I wonder if there is a way to check if the PLC is or not available, can you support me, please?

SmackyP, 2020-07-06:
Instead of using the IsAvailable property, you can ping the ip of the Plc. If you want it built in the Plc class, you can create an extension method

mycroes, 2021-06-04:
`IsAvailable` no longer exists, yay!

closed 2021-06-04 by mycroes

imba-tjd, 2022-06-29:
The Wiki page should be updated https://github.com/S7NetPlus/s7netplus/wiki#check-plc-availability

Either remove the section, or keep it but add a note that it's been removed in 0.10