S7NetPlus/s7netplus #187: Request: Support DATE_AND_TIME complex type

antonsimola, 2018-09-20:
Hello,
I should read DATE_AND_TIME complex type:
http://www.plccenter.cn/Siemens_Step7/Zusammengesetzte_Datentypen.htm
would be nice if the library supported reading these into C# DateTime, or does the library have some way of doing this already?

mycroes, 2018-09-20:
HI @antonsimola, It's not there yet, I'll try to add it right away.

mycroes, 2018-09-21:
Started on it yesterday, got a support call so had to stop... Plan is to add it this evening. Thanks for adding the documentation link BTW, makes life easy!

mycroes, 2018-09-21:
@antonsimola Can you test PR #189? I can't get it to behave _exactly_ as in the documentation, because they're able to set day of week to Thursday for december 25, 1993 (which is a Saturday). I don't consider this a bug. Also, I couldn't find the 4th millisecond digit (again, not a bug on my side :wink:).

antonsimola, 2018-09-21:
Thanks, I will check if I have time to get PLC tomorrow. Latest on Monday I can test. I'm newbie in PLC programming, maybe I'm missing something but I can't find BCD conversion in your commits, is it not needed? Cos I remember reading default 1990-01-01 date from PLC and it was returning for example  for byte[0]=144 (so bcd  /hex 90)

mycroes, 2018-09-22:
> I'm newbie in PLC programming

I don't do PLC programming at all :wink:

> maybe I'm missing something but I can't find BCD conversion in your commits, is it not needed? Cos I remember reading default 1990-01-01 date from PLC and it was returning for example for byte[0]=144 (so bcd /hex 90)

It's definitely not there in the code (although there's no BCD conversion anyway, S7 and x86 store numbers in the same way on byte-level, but I don't do hex translations either). It would be pretty awkward if they would do such a conversion, although I've seen even stranger things in Mitsubishi communication protocols. If it doesn't work, can you provide some samples (actual date time and the 8 bytes representing it) for testing? Would be nice if you could do _December 25 1993, 08:12:34.567_ which is the sample in the documentation, but other samples will do as well of course.

I also couldn't find the date and time type in SAPI-S7 documentation, so I didn't have anything else to verify that the documentation you provided is correct.

mycroes, 2018-09-25:
I see what it's about now. Digits are encoded into 4 bytes per digit, which still seems stupid to me, but I guess there must be some reason for that. I'll update the PR to address this.

mycroes, 2018-09-25:
Updated #189 with BCD support, let me know if it works.

antonsimola, 2018-10-30:
So sorry it has took a while. It seems I cannot get test PLC anytime soon. But I can confirm that my implementation has been in production for a month and it seems fine from Siemens PLC dev environment as well. I code reviewed your stuff and it seems similar. I think this should be merged, maybe with mention of experimental support if necessary.

PR #189 (2019-07-17): Add DateTime support
Fixes #187.
closed 2019-07-17 by mycroes