S7NetPlus/s7netplus #188: S7200 - Connection

JamesHR, 2018-09-20:
Hi, I have an issue trying to connect to S7200.

The S7200 PLC does not have a IP Address, however exist the CP 243-1 module that provides this protocol, but I don't have it.

Is it possible to make the connection using a PPI protocol using a Siemens USB PC adapter?

mkEGTZ, 2018-09-20:
> Hi, I have an issue trying to connect to S7200.
> 
> The S7200 PLC does not have a IP Address, however exist the CP 243-1 module that provides this protocol, but I don't have it.
> 
> Is it possible to make the connection using a PPI protocol using a Siemens USB PC adapter?

----

I´ve been having the same Issue,  some of the information around internet shows that, this kind of PLC is Obsolete.

I hope to solve this ASAP without the CP 243-1.

Does anyone have some information about it?

I´ve been looking forward for any comment.

thnx

mycroes, 2018-09-20:
Hi guys, I'm no expert on Siemens hardware at all, so I had to look up what I could find about the PPI protocol. Unfortunately I could only find a LabView implementation, but I can't read the files (requires LabView, of course) to see what the communication is like. From the filenames I suspect the S7 protocol might be used there as well, which means that it might be possible to apply large parts of the codebase to read/write over PPI (or PPM or MPI for that matter).

If you guys know **any** working program that can use the PPI interface for which the source code is available I can see how much it compares and if it's feasible to add support. I don't have time to do an extensive search for this myself at the moment, but if you can find something I'll check it out.