S7NetPlus/s7netplus #194: Connection issue - Waiting for COTP connect confirm

ccirstea, 2018-10-10:
Hi all,

     I have been using the S7 driver v 0.1.5 until now to connect to a S7-300 PLC. Sometimes, but not all the times, on connect I used to get ErrorCode.IPAddressNotAvailable from Open(). 
     I recently updated to the latest version 0.3.0 and now, sometimes I get an exception with the following message: "Waiting for COTP connect confirm".
     The interesting part with both versions is that the second time I connect it works without issues.
     My workaround for this issue is to handle the exception and try again and yes, it works.

Regards,

mycroes, 2018-10-10:
HI @ccirstea,

That's a strange problem. The relevant code in S7NetPlus is as follows:
```csharp
var response = COTP.TPDU.Read(stream);
if (response.PDUType != 0xd0) //Connect Confirm
{
    throw new WrongNumberOfBytesException("Waiting for COTP connect confirm");
}
```

This means we actually receive data, it just doesn't match what was expected. Do you have this issue with multiple PLC's and different network cards, or just with a single PLC?

I'll create a PR to add the received information to the exception, that way we should be able to get a basic idea of what's happening, that might allow us to fix the issue.

ccirstea, 2018-10-10:
There is just one PLC, a S7-300. I do not have the opportunity to test for other models unfortunately, but for this it does occur.

mycroes, 2021-06-04:
Hi @ccirstea,

Any chance you can retest this against S7NetPlus 0.12.0?

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes