S7NetPlus/s7netplus #195: Reading class from PLC

mycroes, 2018-10-10:
@mycroes I am having issues with a boolean directly connected to a break the beam sensor. The Db class looks like this:

```csharp
    [Location(32)]
    public class PlcRevolverStatusDb
    {
        public bool Initialized { get; set; }
        public bool InputSlotAbsenceState { get; set; }
        public bool OutputSlotAbsenceState { get; set; }
        public bool ProcessingAreaAbsenceState { get; set; }
        public ushort DummyValue { get; set; }
    }
```

ProcessingAreaAbsenceState directly connected to the sensor sometimes reads opposite to the presence. Could the ushort influence this faulty read?

_Originally posted by @ccirstea in https://github.com/S7NetPlus/s7netplus/issues/135#issuecomment-428620299_

ccirstea, 2018-10-10:
I read the register and others each 100 ms

```
private void Work()
        {
            while (!mStopThreadCondition)
            {
                try
                { 
                    if (mCommClient.IsConnected)
                    {
                             // other registers, same logic
                            PlcRevolverStatusDb Db = new PlcRevolverStatusDb();
                            int plcDbIndex = (typeof(PlcRevolverStatusDb).GetCustomAttributes(typeof(LocationAttribute), true).FirstOrDefault() as LocationAttribute).Location;
                            mCommClient.ReadClass(Db, plcDbIndex);
                            mRevolverStatusRegister.SetNewRegisterValue(Db);
                   }
                }
                catch(Exception ex)
                {
                        Log ... 
                 }
                }
                //thread sleep
                Thread.Sleep(100);
            }
        }
```

where mCommClient is a wrapper over S7Net

`public class Sharp7Client
    {
        private Plc _plc = null;
        private object _lock = new object();

        public bool IsAvailable
        {
            get { return _plc.IsAvailable; }
        }

        public bool IsConnected
        {
            get { return _plc.IsConnected; }
        }

        public Sharp7Client(string ip, short rack, short slot)
        {
            _plc = new Plc(CpuType.S7300, ip, rack, slot);
        }

        public ErrorCode Connect()
        {
            try
            {
                if (_plc.IsAvailable)
                {
                    _plc.Open();

                    if (_plc.IsConnected)
                        return ErrorCode.NoError;
                    else
                        return ErrorCode.ConnectionError;
                }
                else
                    return ErrorCode.IPAddressNotAvailable;
                
            }
            catch(PlcException plcException)
            {
                if(plcException.Message.Contains("Waiting for COTP connect confirm"))
                    return ErrorCode.IPAddressNotAvailable;
                return plcException.ErrorCode;
            }
        }

        public void Disconnect()
        {
            if (_plc.IsConnected == true)
                _plc.Close();
        }

        public int ReadClass(object sourceClass, int db, int startByteAdr = 0)
        {
            lock(_lock)
            {
                return _plc.ReadClass(sourceClass, db, startByteAdr);
            }
        }

        public ErrorCode WriteClass(object classValue, int db, int startByteAdr = 0)
        {
            lock (_lock)
            {
                try
                {
                    _plc.WriteClass(classValue, db, startByteAdr);
                    return ErrorCode.NoError;
                }
                catch(PlcException plcException)
                {
                    return plcException.ErrorCode;
                }
            }
        }
    }`

ccirstea, 2018-10-15:
@mycroes . I understand that there are issues with ReadClass, so my question is, which is the most reliable way of reading a DB from the PLC using S7 in your experience?

Thank you,

Mallowan, 2018-10-20:
> 
> 
> I understand that there are issues with ReadClass, so my question is, which is the most reliable way of reading a DB from the PLC using S7 in your experience?
> 
> Thank you,

Hi there,
what about to show screenshot of other side? I mean DB that you are trying to read.

Mallowan

keeepcool, 2018-10-31:
Good evening,

I'm also having a little problem with writing a class, readClass works fine and all the values match, but writeClass will hang my program even with a try catch block, what is interesting is that the values are updated in the PLC, but writeClass never returns and so the program must be terminated.