S7NetPlus/s7netplus #197: WriteClass never returns

mycroes, 2018-10-31:
Good evening,

I'm also having a little problem with writing a class, readClass works fine and all the values match, but writeClass will hang my program even with a try catch block, what is interesting is that the values are updated in the PLC, but writeClass never returns and so the program must be terminated.

_Originally posted by @keeepcool in https://github.com/S7NetPlus/s7netplus/issues/195#issuecomment-434725052_

mycroes, 2018-10-31:
@keeepcool Please provide a minimal sample of the failing program. You can also do that as a separate GitHub repository, I'll take a look as soon as the information is here.

barnabaskisstdk, 2020-10-23:
Hi,

I have a same problem, and the WriteClass can't handle Lreal datatype.
Thanks in advance

closed 2021-06-06 by mycroes