S7NetPlus/s7netplus #198: Reading multiple vars and a string with odd length fails

bauerju, 2018-11-08:
Hi 
When I read multiple vars and one of the vars is a StringEX with an odd legth, I get an exception: 
 PLC communication failed with error WrongNumberReceivedBytes

But this only occurs if the string is not the last item to read.  

I found the problem in class Plc (file PlcHelpers.cs) in method **ParseDataIntoDataItems**
The offset caculation is only adjusted for vartype byte or bit. I added String and StringEX.  

                // Fill byte in response when bytecount is odd
                if (dataItem.Count % 2 != 0 && (dataItem.VarType == VarType.Byte || dataItem.VarType == VarType.Bit 
                                             || dataItem.VarType == VarType.String || dataItem.VarType == VarType.StringEx ))
                    offset++;

Jürgen

mycroes, 2021-06-06:
Still needs to be fixed, relevant code location is PLCHelpers.cs line 246 (reminder for self).

scamille, 2021-06-06:
Ok I found out it just needs to be always aligned to 2, which you also have implemented like this at https://github.com/mycroes/Sally7/blob/812f97064e42a12bbdf49d1fda1b975d48810ac1/Sally7/S7ConnectionHelpers.cs#L171

I will fix this.

mycroes, 2021-06-06:
@scamille Didn't see this message yet, but really nice how you're fixing this with code I've written somewhere else 😜 

@bauerju It's taken a while, but the fix will be part of the next release.

PR #397 (2021-06-06): Fix ReadMultipleVars alignment bug
This PR tries to fix #198
closed 2021-06-06 by mycroes