S7NetPlus/s7netplus #200: Why gc.collect necessary when reading s7-300?

349516385, 2018-12-14:
My code like this:

In a winform timer of cycle 35ms
{
        Plc plc = new Plc(CpuType.S7300,"192.168.1.1",0,2);
        plc.Open;
        plc.ReadStructure;  // omitted the parameters 
        plc.Close;
        //  plc = null;   
       // GC.collect();
}
 
I found than I can read the plc  74 times at most without "GC.collect".
After 74 timers  it will report errors: TPKT is not connected (almost like this).
 I tried lots of methods and I must add the last two sentence to avoid the error.
Can anyone explain this? Thanks a lot!

mycroes, 2018-12-14:
GC.Collect() shouldn't be necessary at all. My guess is some other part of the code is causing issues, but without the code I can't be sure. If you can provide the code (preferably as a GitHub repository) I will take a look. It would also help if you'd put the actual exception message in the issue (the message you gave is a mix of exceptions which could probably point to half of the exceptions in the project).

allatrue, 2018-12-15:
So you create every 35 ms a new instance and make a new connection to PLC? Why not to reuse one instance?

mycroes, 2019-05-27:
Closing due to missing feedback.

closed 2019-05-27 by mycroes