S7NetPlus/s7netplus #204: TPKT is incomplete / invalid exception

piayping, 2019-01-16:
I set up three timers, two for writing and one for reading. When I read and write S7-1200 at the same time, many “TPKT is incomplete / invalid” exceptions are thrown，Almost all read actions throw  exception

mycroes, 2019-01-16:
Connections are not suitable for concurrent use. If you want to read and write from multiple threads (which you probably get from the multiple timers) you need to use a sychronization mechanism (locking, queing).

piayping, 2019-01-19:
Thanks a lot for your advising! I use ConcurrentQueue and Synchronization to solve my problem.

closed 2019-01-19 by mycroes

pahager, 2020-10-26:
Hi, can you describe please how to use  ConcurrentQueue and Synchronization? I'm trying to read value of DB in one Thread and writing in another Thread and getting the same error as you: TPKT is incomplete / invalid exception. How to fix it?

peymanmajidi, 2021-01-21:
What does TPKT stand for?