S7NetPlus/s7netplus #207: How to use Relection to let function “ReadStruct” to support array

tsheyu, 2019-02-09:
Hi Michael:
          I am a Rookie in C# Programming, Right now I have a problem I want function "S7.net\Struct.cs\GetStructSize“ to support array data. So I rewrite the function as "GetStructSize<T>(T data)", because only in this way I can get the length of the array member.  I have a problem to invoke Generic function using reflection. Could you have a look in the default line?
 ```
   public static int GetStructSize<T>(T data)
        {
            double numBytes = 0.0;
            var infos = data.GetType().GetFields();
            foreach (var info in infos)
            {

                switch (info.FieldType.Name)
                {
                    case "Boolean":
                        numBytes += 0.125;
                        break;
                    case "Byte":
                        numBytes = Math.Ceiling(numBytes);
                        numBytes++;
                        break;
                    case "Int16":
                    case "UInt16":
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        numBytes += 2;
                        break;
                    case "Int32":
                    case "UInt32":
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        numBytes += 4;
                        break;
                    case "Single":
                    case "Double":
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        numBytes += 4;
                        break;
                    case "DateTime":
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        numBytes += 8;
                        break;
                    default:
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        if (info.FieldType.IsArray)
                        {
                            Type individualType = info.FieldType.GetElementType();
                            TypedReference dataReference = __makeref(data);
                            var _arrayType = info.GetValueDirect(dataReference) as Array;
                            int length = _arrayType.Length;
                            /// To Invoke self function to calculate the length, I do not know how
                        }
                        else
                            numBytes += GetStructSize(info.FieldType);
                        break;
                }
            }
            return (int)numBytes;
        }
```

mycroes, 2019-03-02:
HI @tsheyu,

The issue with arrays is that they're not of fixed length (in C#). The closest you can get to defining sizes for arrays is by using attributes, those could then be read out by S7NetPlus as well to use the desired size.

For now you can use DataItems to manually mimick the struct you have in PLC while retaining the ability to read multiple items in one go.

tsheyu, 2019-03-07:
Hi mycroes:
Attribute is OK, but the initialization still needs declaration and the code will be unsafe.
Two weeks ago, I make do by using "class" instead of "struct" to solve this issue. although more codes have to be written for class initialization, but the code is managed so I think it is better to use class

closed 2019-03-20 by tsheyu