S7NetPlus/s7netplus #208: I have problems connecting to an S7 200

frustis, 2019-02-20:
My code:

`using System;
using System.Collections.Generic;
using System.Linq;
using System.Threading.Tasks;
using System.Windows.Forms;
using S7.Net;

namespace WindowsFormsApp1
{
    static class Program
    {
        /// <summary>
        /// Punto de entrada principal para la aplicación.
        /// </summary>
        [STAThread]
        static void Main()
        {
            // Application.EnableVisualStyles();
            Plc plc = new Plc(CpuType.S7300, 10.127.85.210, 0, 1);
            /// plc.
            plc.Open();

            Console.WriteLine(plc.IsAvailable);
            Console.WriteLine(plc.IsConnected);
            // var dwords = plc.Read(DataType.DataBlock, 1, 0, VarType.DWord, 20);
            // ushort result = (ushort)plc.Read("DB1.DBW0");

        }
    }
}`

In the plc.Open () I get an exception:

Excepción producida: 'S7.Net.PlcException' en S7.Net.dll
Excepción no controlada del tipo 'S7.Net.PlcException' en S7.Net.dll
Couldn't establish the connection to 10.127.85.210.
Message: TPKT is incomplete / invalid

Any Idea?

Thanks

mycroes, 2019-03-02:
Hi @frustis,

At first I saw the PLC type in your code didn't match the issue title, but since you corrected that I'm assuming you are using the right values now. Unfortunately your issue is hard to debug. The PLC refuses the connection, the reason is unclear. I'm guessing it has to do with some data in the Connection Request as sent by S7NetPlus, but what part I don't know. There are more open issues related to this, I'm trying to improve the situation but I don't have much time these days... I will try to get back on this soon. In the meantime, have you tried any other library for S7 communication to see if it works?

mycroes, 2020-09-08:
Just spotted the `plc.IsConnected` usage, unfortunately that kills the connection. Please retest this without `plc.IsConnected` and preferably against a newer S7NetPlus version.

scamille, 2020-09-09:
IsConnected does not kill your connection I believe, only IsAvailable might.

The main problem here should be fixed in version 0.7.0, so just try again with the latest version please.

mycroes, 2021-06-04:
Also, this would probably be fixed by #396 when you specify TSAP's manually, because from what I understood the S7 200 doesn't support a fixed set of TSAP addresses as most newer PLC's do...

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes