S7NetPlus/s7netplus #210: Wrong size returned by Struct.GetStructSize on some structs

rashandan, 2019-03-06:
In example:
`struct Bridge 
{
  public float float1; 
  public bool bool1;
  public bool bool2;
  public bool bool3;
  public bool bool4;
}`
size of abowe structure must be 5, but when reading by ReadStruct, it spawns exeption.
Inside this function, call to Struct.GetStructSize returns 4.
I believe there must be something like `return (int)Math.Ceiling(numBytes);` instead of `return (int)numBytes;`
This happens when struct ends by from 1 to 7 bool-s.

//sorry for my English

tsheyu, 2019-03-07:
I also find this problem before. Since in my program scenario, the struct contains struct. So the size must be Even -aligned. 
In your case  the size should be 6 bytes. U can open the data block to check the real size.
 
If it is confirmed, At the end of the function  "Struct.GetStructSize" add the code like
` numBytes = Math.Ceiling(numBytes);
                    if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                        numBytes++;
                    break;`

mabc666, 2021-04-17:
the author round the size of struct it looks like a bug and hasn't been fixed  yet.
you can round up the size of struct 
`return (int)Math.Ceiling(numBytes); `
It can slove this problem