S7NetPlus/s7netplus #213: S7-300 Writing value to Datatype Memory set value to zero

kivannc, 2019-03-31:
I can read the values but when I try to write a new integer value it always set the value to zero in the plc. I tried both write methods. Address is MW250

mycroes, 2019-04-02:
Hi @kivannc,

Could you also show the part where you're reading the values? I personally have no experience with writing memory datatype, so I'd have to look in the code for reasons why it wouldn't work.

kivannc, 2019-04-02:
After lots of try and searching I finally figured it out. You need to cast to value to UInt16. Then it started writing to plc

`var convertedValue = Convert.**ToUInt16**(valueFromAbove);`
`Instance.Write(DataType.Memory, 0, 250, convertedValue);`

closed 2019-04-02 by kivannc