S7NetPlus/s7netplus #215: Sometimes Value is not written to DB

tomaskrutak, 2019-05-17:
Hi,
I am connected to PLC and writing to DB using method :
_S7Plc.Write("DB100.DBW0", 1);

Write function works most of the time but once in around 6 write attempts write fails. So DB value is not changed.
Do I miss some PLC setting or what can be reason?
Thank you
Tomas Krutak

mycroes, 2019-05-17:
Hi @tomaskrutak,

I have experienced similar issues on multiple occassions (with Sally7 and Siemens' own SAPI-S7, I don't actually use S7NetPlus myself). There is some datablock property specifying data direction. I believe by default it's set to in/out, which causes this issue. I'm sorry this is a bit vague, I'm not a PLC programmer. Let me know if you can find the setting, otherwise I can ask the PLC programmers at my company.

tomaskrutak, 2019-05-20:
Hello,
Thank you, but Im not aware there is such setting on shared DB if its in or out...

> Od: "Michael Croes" <notifications@github.com>
> Komu: "S7NetPlus/s7netplus" <s7netplus@noreply.github.com>
> Datum: 17.05.2019 21:53
Hi @tomaskrutak <https://github.com/tomaskrutak>,
I have experienced similar issues on multiple occassions (with Sally7 and Siemens' own SAPI-S7, I don't actually use S7NetPlus myself). There is some datablock property specifying data direction. I believe by default it's set to in/out, which causes this issue. I'm sorry this is a bit vague, I'm not a PLC programmer. Let me know if you can find the setting, otherwise I can ask the PLC programmers at my company.

mycroes, 2019-05-20:
Hi @tomaskrutak,

In the attached image you can see variable *IN1* and *IN2* declared specifically as *Input*. If they would have been declared as *InOut* that often results in the written values not received in the PLC.

Regards,
Michael

[image]

mycroes, 2019-05-27:
Closing due to missing response. Feel free to reopen if the issue persists.

closed 2019-05-27 by mycroes