S7NetPlus/s7netplus #216: Real industry application on s71214

simotion1, 2019-05-20:
Dear all;
I downloaded last version of s7netplus from nuget. We developed an application reading and writing to an s71214 on a real industrial application 24h 7 days so we need that is reliable.
Does anyone use s7netplus on real industry?
Is it completely open source? 

We solved the open() delay, by opening a new thread every time that we use open() method.

Thanks

mycroes, 2019-05-20:
Hi @simotion1,

It should definitely work. Also, it is fully open source. On the other hand I must admit I don't use S7NetPlus (must sound awkward since I'm the current maintainer). I developed [Sally7](/mycroes/Sally7) to my own needs, which actually is industrial 24/7 applications. The underlying principles in both libraries are roughly the same. S7NetPlus supports more features but has a more convoluted API, Sally7 is faster.

I'd say pick whatever library best suits your needs, both should work fine.

FalcoGoodbody, 2019-05-22:
Hi all,

I am using S7NetPlus since the beginning of this year to log about 130 machine parameters from a S7-1500 PLC into a MSSQL-Database. I am planning to connect nearly all PLCs from our production plant, which are round about 10 pcs.
Until now I did make really good experince with S7NetPlus and it works from the first day absolutely stable and reliable 24/7.

mycroes, 2019-05-27:
I think this answers the question, feel free to reopen if it doesn't.

closed 2019-05-27 by mycroes