S7NetPlus/s7netplus #221: Read struct with array(s)

Styxer, 2019-06-10:
I have noticed `ReadStruct` using a function `GetStructSize` which in turn uses cases for each type 
but i do not see there a case for an array or any other complex type(s)

my class looks like [this ](https://ibb.co/0VwFNtw) how can i achieve reading VCM_Setpoint?

this is my struct in code(i always get **null** on VCM_Setpoint, all other values have no problem while trying to read them)
```
    public struct CIP_B
        {

           public bool Start_Req;
           public bool Start_Ack;
           public bool End_Req;
           public bool End_Ack;
           public bool Ended_OK;
           public bool Step_Interupted;
           public bool Value_Changed;
           public Byte CIP_OBJ;
           public Byte CIP_TYPE;
           public ushort[] VCM_Setpoint;
           public Int16 Step_Num;
           public Int16 changedValue;
           public Byte[] Spare;

           public CIP_B(int size = 16)
           {
               Start_Req = false;
               Start_Ack = false;
               End_Req = false;
               End_Ack = false;
               Ended_OK = false;
               Step_Interupted = false;
               Value_Changed = false;
               CIP_OBJ = 0;
               CIP_TYPE = 0;
               VCM_Setpoint = new ushort[16];
               Step_Num = 0;
               changedValue = 0;
               Spare = new Byte[10];
           }
}
```

mycroes, 2021-06-04:
Hi @Styxer,

Sorry for my late reply. There's no good reliable way to support arrays in structs or classes because their length isn't fixed in C#. It's a feature that has been requested occasionally, it's not one I'm very eager to implement...

Regards,
Michael

Equino3108, 2021-11-07:
Not sure if this is still up to date but when I was working with s7netplus about a year ago, I ran into the same problem and was able to get it done using an attribute class to define a "pseudo-fixed" array length. This approach works for strings as well, since a string in c# is nothing else but a char array without a fixed length.

Just add an attribute class to your struct definition

[ArraySize(16)]
public ushort[] VCM_Setpoint

Then in the read struct, as far as I remember there are classes to "translate" each type to s7 reading instructions. Just add a new class there for arrays and use reflection to detect the Array type, pass the attribute class length parameter in to read ushort or char or whatever length times.

I sadly don't have access to the code I wrote back then anymore, since this was a contract work, but it's not very complicated at all. You already got all the different read/write methods for each elementary type, with the attribute class you can just add a loop around those to read it x time (x being the ArraySize).

scamille, 2021-11-07:
I don't see why this couldn't be done, as long as the array object is already initialized when passed to s7netplus you should be able to get the size easily.

What you can't do is let s7netplus infer the size based on the  data in the plc itself.