S7NetPlus/s7netplus #224: S7300 "WrongNumberReceivedBytes"

valesQQ, 2019-06-25:
Hello everyone. I'm doing an internship and I have to read data from a PLC using a C# application.
It's a S7-300 connected to a WINLC RTX.
I'm able to connect to the PLC but everytime I try to read some data (using ReadStruct, ReadBytes and simply Read) I get the same error "WrongNumberReceivedBytes".
The data I try to access is located inside a Struct itself inside an UTD itself inside a DB.
The code in itself is very simple, it's mostly the functions in the wiki with some code to log data in a .txt file.
Some example of functions:

var data = (bool)plcSiemens.Read("DB1110.DBX140.0");
var bytes = plcSiemens.ReadBytes(DataType.DataBlock, 1110, 140, 1);

If someone had an idea where the error could come from , it would help me a lot !

FalcoGoodbody, 2019-06-26:
Hello valesQQ,

I can check this behaviour here at a running machine an will give feedback when its done.
But I do not have a S7-300, I have S7-1500.

I also use S7.Net an I read a lot of values. For the situation you descriped I am using Read.Class and have no problems so far. Maybe you can try it that way.

FalcoGoodbody, 2019-06-26:
So I have tested it with a S7-1515-2PN.
It was the same constellation. A struct within an udt wich is declared within a DB.
I used the following code to read the single values, but would prefer to read the whole struct or even better as class:

` Glatt2TestINT = ((ushort)plcGlatt2.Read("DB76.DBW2")).ConvertToShort();
                    Glatt2TestREAL = ((uint)plcGlatt2.Read("DB76.DBD4")).ConvertToFloat();
                    var statusTest = plcGlatt2.ReadBytes(DataType.DataBlock, 76, 0, 1);
                    Glatt2TestBool = statusTest[0].SelectBit(0);
`

valesQQ, 2019-06-26:
Okay, thank you.
It may be a hardware problem so if this code is working on your side.

I plan to read the whole struct but first I need to get a working reading so I'm only focusing on one data for now.
I will have a look at Read.Class also.

FalcoGoodbody, 2019-06-26:
@valesQQ  check your code reading one bit. When you comare it to my code you will see that I first read one byte and then select one bit from this byte. This should work fine.
`var statusTest = plcGlatt2.ReadBytes(DataType.DataBlock, 76, 0, 1);
Glatt2TestBool = statusTest[0].SelectBit(0);`

mycroes, 2019-06-26:
Hi @valesQQ, sorry for my late response. I hope you can get started with the answers already given, but I'd be happy to help out (just a little short on time). You could alternatively try if [Sally7](/mycroes/Sally7) works, any feature working there and not in S7NetPlus I'd be willing to fix.

valesQQ, 2019-06-26:
Ok, thank you guys, I will update if I get somethig new.

valesQQ, 2019-06-27:
Hey,
I've got it working. The problem was that I tried to read data from a different PLC than the one I was connected to...

Still a lot to learn. Thank you for the test @FalcoGoodBoy, made me realize that the issue was probably not in the code.

closed 2019-06-27 by valesQQ