S7NetPlus/s7netplus #228: S7-1200 read/write error

allibragi, 2019-08-07:
Hi, I'm unable to read data from a S7-1200.
This is the code I'm using, in the comments the types of errors:
```

Plc _S71200 = new Plc(CpuType.S71200, "192.168.0.1", 0, 0);
_S71200.Open();

if (_S71200.IsConnected)
{
 bool exit= (bool)_S71200.Read("DB100.DBX0.0"); //IndexOutOfRange
 //or
 DB100 db = new DB100();
 _S71200.ReadClass(db, 100); //IndexOutOfRange
//or
 var bytes = _S71200.ReadBytes(DataType.DataBlock, 100, 0, 200);//Unable to read data: connection interrupted by remote host.
}
```
I've tried to write and I've got again the IndexOutOfRange exception.
This is my first 1200 but is a simple code that runs with other PLCs that I've worked with.

Thanks in advance

mycroes, 2019-08-07:
Hi @allibragi,

I can't diagnose where the exception is coming from just by the type. However, you're almost certainly passing wrong values for rack and slot. I think `slot` should probably be `1`, not entirely sure though. Let me know how that works out for you.

allibragi, 2019-08-08:
Hi @mycroes, thank you for the support.
I've found a documentation which says that you can use both slot 0 or 1. I've tried both but i got the same error.
I will try to reference directly the source code solution and debug it.

mycroes, 2019-08-15:
HI @allibragi,

Somehow I missed your update comment. What I'm guessing is that the PLC is returning an error. There's not much error handling going on in S7NetPlus with regards to variable read/writes. Best thing you can do right now is try with [Sally7](/mycroes/Sally7), it'll probably give you some sort of error. If you can copy the contents of s7data when you get the exception and paste them here I might also be able to figure what's going on.

Regards,
Michael

allibragi, 2019-08-16:
Hi @mycroes, thank you again for the support.

I've tried Sally7, I've got this error while trying to read the data:
`"The format string can only be \" G \ ", \" g \ ", \" X \ ", \" x \ ", \" F \ ", \" f \ ", \" D \ "o \ "d \"."`

tobiasSM, 2019-08-16:
> ### Update:
> 
> For all the three methods (Read, ReadClass, ReadBytes) the program stops in the file PlcSynchronous.cs, ReadBytesWithSingleRequest method, line 380:
> 
> ```
> if (s7data == null || s7data[14] != 0xff)
>  throw new Exception(ErrorCode.WrongNumberReceivedBytes.ToString());
> ```
> 
> In the "if" it search s7data[14] but my s7data is a byte[12].

I got the same error today. I cant look into the issue any further, but is it possible that it is a problem on the S7-1500 side because of a onfiguration error?

mycroes, 2019-08-18:
HI @allibragi, @tobiasSM,

I messed up the last Sally7 version, so when it detected an error it would throw some useless exception. I just pushed Sally7 0.7.1 to nuget.org, which should fix that error and hopefully provide a more meaningful error.

As for what @tobiasSM said about PLC-side configuration, yes, that's probably it. The PLC returns only 12 bytes because it returns an error. I don't know what the error is without seeing the data. There's no detection/handling for those errors in S7NetPlus and adding it would probably be quite some work because there's a lack of structure for handling responses.

So anyway, if you guys can either test Sally7 or show the received bytes (`s7data`) I can probably figure out what's going wrong.

Regards,
Michael

tobiasSM, 2019-08-19:
I was told the "the put and get where activated" (whatever that means) in the configuration of the S7 and now my error is gone.

allibragi, 2019-08-19:
With the new version I got this error:
```
An error was returned during communication:
	Message type: Ack / 0x02
	Error class: ApplicationRelationShipError / 0x81
	Error code: 0x4
	Combined error: 0x8104
```
I was unable to find the s7data tag in the error properties. Unfortunately I cant access to the s7 right now, I will talk to the techniciat that prapared it to check the option @tobiasSM told.

jorgeandres6, 2019-09-02:
Hey everyone,

I had the same problem, I used wireshark in order to look up for the problem and I found out that S7COMM protocol send an error (0x8104 - Function not suported, or with S7-1200/1500 "Permit access with PUT/GET communication from remote partner" needs to be enable). Even tough this PUT/GET option was checked, it seems it wasn't workin, so I just wrote a new simple project at TIA portal, just with a datablock with an only bool variable in it and obviously all the configurations needed in order for the library to work propertly, downloaded it to the PLC S7-1200, and guess what, IT WORKED!!! so y returned to my original project, downloaded it again in the PLC, and it started working just fine :)

Hope this will be usefull for you @allibragi

allibragi, 2019-09-30:
Sorry for the waiting, the technitian showed up eventually.
The issue was the same of @jorgeandres6, he reuploaded the project and now it works.
Thank you all for the help.

closed 2019-10-03 by allibragi