S7NetPlus/s7netplus #230: Connect to LOGO

ChermyaninAA, 2019-08-15:
Hey. I want to connect to LOGO. Do not tell me which rack and slot to use to create an instance of the PLC class.
I would be grateful if you show an example of code for connecting to the LOGO.

mycroes, 2019-08-15:
Hi @ChermyaninAA,

I'm not sure what you mean by _Do not tell me which rack and slot to use_, but I'll just assume it's meant in some friendly way. I created a gist just for you which should get you going here:

https://gist.github.com/mycroes/9bdc7cffdef08943edea1918812706a6

Let me know how this works out for you.

Regards,
Michael

ChermyaninAA, 2019-08-16:
Thank you, this is what you need!
And another question: logo0ba8 are no longer supplied and replaced with logo0ba0, is your library compatible with them?
[model comparison](https://mall.industry.siemens.com/goos/catalog/Pages/mmpdata.ashx?lang=ru&MLFB1=6ED1052-2MD00-0BA8&MLFB2=6ED1052-2MD08-0BA0&)

mycroes, 2019-08-16:
I don't know for sure, but I think it is. We might need to update the id's. Let me know if it works, if it doesn't I'll try to fix it.

ChermyaninAA, 2019-08-26:
Hey. Everything works with logo0ba0. thanks for the help

closed 2021-06-04 by mycroes