S7NetPlus/s7netplus #231: Multiple Writes via PC Application

Crane49, 2019-08-15:
Anyone,

I have a PC application running as a serivce times 3. It is polling the S7-1518 via Ethernet with only 2 hops in between running in a vm space. They are reading about 30 bytes of data every 50ms. I am also writing data from each the threee locations to separate registers in the PLC every 1.5 seconds. In total 1 TASP connection  with 1 read & 4 writes per PC Appplication service. Every once in awhile I get a failure where the data is not written to the PLC without any errors from the S7NetPlus. Is there any way to gain greater visability of the actual transactions per service via the S7NetPlus driver? I am currently looing for a way to scan these connections via the PLC. I am also proposing separating the connecions via a Cp1543-1 since I also have multiple Open user conneciotns and large HMI(OPC) on the same communcation port with the PC based services connected to the PLC via X3.

Any insight would be greatly appreciated!
@Crane49

mycroes, 2019-08-15:
HI @Crane49,

I'm guessing what you're doing is along the following lines:
```csharp
var plc = new Plc(CpuType.S71500, ip, 0, 0);
await plc.OpenAsync();
await plc.ReadMultipleVarsAsync(dataItems);
plc.Close();
```

I'm slightly amazed by the fact you're saying you get no errors, that doesn't make much sense. However, if your code is as above, I can tell you there's a fair chance it doesn't work (but will give you errors) because the connection ID's used by S7NetPlus can't be used for multiple connections to the PLC (at least, that's my experience with a 1512 or 1516 iirc).

One other option is that you're have the variables declared as _InOut_, that seems to be causing intermittent write failures as well (see #215).

Regards,
Michael

Crane49, 2019-08-20:
Hi Michael,

Thanks for the response. I am not using DataBlocks with In/Out Properties. I have however gotten it to work as of now.

I added an CP1543-1 Communication Module and Update its firmware to v2.1.4 as well as increase my processor slice time/communication load to 40% after looking to see that the USer comunication load was higher than the projects. This seems to have at this time since things seem to change resolve my write varible issue.

Some clarification on the Setup from Above we running 3 VB/ applicaions via  servioce on VM machine talking to the S7-1518-4 processor via a CP1543-1 comunication module. I have 1 central OPC UA Hmi reading over 20,000 tags & 2 Open User Connections (Sockets). I now have 13 additional connections via S7-Transport (S7NetPlus) coming in doing reading and writing. We read every 50ms and write every 1.5s worst case. Now the transports connections will grow as we have one line offline for repairs.

Hope This Helps anyone,
Thanks,
Mike C. (Crane49)

mycroes, 2021-06-06:
Hi @Crane49,

I was wondering if you're still affected by this issue? For one, multiple connections using the same TSAP's won't work, so to do that you'd have to setup additional connections in TIA yourself and use those ID's, that does require the changes in #396 as well though...