S7NetPlus/s7netplus #232: Problems reading merker bit

YastrebovV, 2019-08-28:
Hello to all! I found a small bug)
It was: `  case "M":
                                // Memory
                                dataType = DataType.Memory;
                                varType = VarType.Byte;
                                break;`

Fixed: `  case "M":
                                // Memory
                                dataType = DataType.Memory;
                                varType = VarType.Bit;
                                break;`

allibragi, 2019-10-03:
Hi @YastrebovV, i think you should submit a pull request with the bugfix. 
It's easer for the devs to check it and put it in the official release.

mycroes, 2021-06-06:
Has been fixed for quite a while, closing.

closed 2021-06-06 by mycroes