S7NetPlus/s7netplus #233: Index was outside the bounds of the array

jorgeandres6, 2019-09-02:
Saludos,

Al tratar de hacer una simple operacion de lectura me indica el error que se muestra en la imagen. He seguido los pasos del manual fielemente, he configurado mi PLC (S7-1200) adecuadamente, tengo una conexión exitosa y lo verifico a través del metodo IsConected, sin embargo no puedo ni leer ni escribir en ninguna variable.
Agradezco su ayuda.

Greetings,

Im trying to read and write on my PLC (S7 - 1200) varibles but I only obtain the message that is shown in the image attached. I followed the instructions from the manual step by step but it still not working.

Please some help. thanks.

[image]

[image]

allibragi, 2019-10-03:
Do you have the same error using other adresses?
Like `plc.Read("DB1.DBX14.4")` or `plc.Read("DB1.DBX0.4")`

sametdumankaya, 2020-02-18:
I'm having same issue here. I tried to read directly from the variable name and that operation threw an "Index was outside the bounds of the array" exception as well. Can't read neither with bytes or with variable name.

sametdumankaya, 2020-02-18:
> I'm having same issue here. I tried to read directly from the variable name and that operation threw an "Index was outside the bounds of the array" exception as well. Can't read neither with bytes or with variable name.

I solved it. I forgot to permit PUT/GET operations on PLC. When I check the checkbox, I was able to read and write.

[image]

mycroes, 2020-09-08:
Please retest against 0.7.0, in worst case it at least provides more meaningful exceptions.

mycroes, 2021-06-06:
Closing due to missing feedback, feel free to reopen if required.

closed 2021-06-06 by mycroes

OlivierMiracle, 2023-03-07:
Hi!
I am trying to get this to work, but I am getting the same exception. There are the steps i took:

- Create a global DB [DB1]
- Unchecked "Optimized block access" in its properties
- In properties of PLC (Simatic S7-1200) I have set full access level
- I have checked PUT-GET communication

Next, I have created a few unit test methods:

For checking avaibility:
```
[Fact]
public void PlcRead_IsAvailable()
{
    //Arrange
    Plc plc = new Plc(CpuType.S71200, "192.168.0.1", 0, 0);

    //Act
    bool result = plc.IsAvailable;

    //Assert
    Assert.True(result);
}
```

For checking connection:
```
[Fact]
public void PlcRead_IsConnected()
{
    //Arrange
    Plc plc = new Plc(CpuType.S71200, "192.168.0.1", 0, 0);

    //Act
    plc.Open();
    bool result = plc.IsConnected;
    plc.Close();

    //Assert
    Assert.True(result);
}
```

And, finally, for write and read instructions.
```
[Fact]
public void PlcRead_WriteAndRead()
{
    //Arrange
    Plc plc = new Plc(CpuType.S71200, "192.168.0.1", 0, 0);
    byte expected = 10;
    byte result;

    //Act
    plc.Open();
    plc.Write("DB1.DBB0", expected);

    result = (byte)plc.Read("DB1.DBB0");

    //Assert
    Assert.Equal(expected, result);
}
```

Only the last one is failing with exception:
```
S7.Net.PlcException : Index was outside the bounds of the array.
---- System.IndexOutOfRangeException : Index was outside the bounds of the array.
```

The problem doesn't not depend on type of variables, their number, casting etc.