S7NetPlus/s7netplus #234: Any way to add DateTime and string support to ReadStruct / ReadClass?

Equino3108, 2019-09-10:
I realized strings and datetime are not supported in ReadStruct / ReadClass Methods.
Is there a way to get a datetime field into These structs/classes ? My first idea was to try and read them as strings, but since they aren't supported either?

Thanks

FalcoGoodbody, 2020-09-09:
Hi @Equino3108,

it is possible to read strings from a plc. Check the issue #239 . There is an example provided.

tolgacakir, 2021-02-09:
Hi,
`ReadStruct`/ `ReadClass` methods do not support **strings**, it should be written to the **document** or **wiki** page.
Will `ReadStruct`/ `ReadClass` methods support **strings**?

scamille, 2021-02-09:
#365 would also enable support for string in struct/classes. Note that this requires adding annotating your string type and capacity, since that can't be inferred from your string alone.

Datetime support in struct/class would also require some annotations to tell S7NetPlus which S7 datetime type to use (there are multiple). And anyway, someone would have to implement it.

You can of course use the same code as @FalcoGoodbody provided just with VarType.DateTime and then cast to (DateTime).

mycroes, 2021-03-12:
I'm leaving this issue open for now (#365 will address the string part when merged), there's no reason why DateTime couldn't be implemented in a similar fashion to how #365 implements string reading (except for the fact that it's simply not necessary to specify a length for DateTime).