S7NetPlus/s7netplus #235: Bug in Struct.GetStructSize() ?

Equino3108, 2019-09-10:
I got a struct with size 16 Bytes and at the end, I also define 4 bool(bit) fields, which brings size up to 16.5 as GetStructSize adds 0.125 per bool field it finds.

At the end of GetStructSize, you return (int)numbytes which cuts off the .5 and makes the method return a size of 16.  After that it crashes at some point in Struct.ToBytes() - apparently because it tries to write bit values to byte 17 which isn't present i the bytes[] array.

I believe GetStructSize() should return (int)Math.Ceiling(numBytes) instead.

mycroes, 2020-09-08:
Sorry for leaving this without response for so long, but yes, you're absolutely right about this. I'm not sure if it should round to an even number of bytes as well, but I think it might. Marking this as a reminder, let me know if this has high priority for you.

scamille, 2020-09-09:
As far as I know a UDT also always starts at an alignment 2 bytes, or in other words at an even address.
The same is true for for data types with size 2 or more, at least for S7-1500. 

So assuming the C# struct tries to mimic the S7 one without the need for filler bytes, we should also handle that.

Probably does not happen that often since you PLC programmers don't like to waste bytes or even bits :-P

FalcoGoodbody, 2020-09-09:
Correct UDTs always starts at an alignment 2 bytes because S7 works word-oriented. Also variables within an datablock from type word, int, dint and real always start at even address.

scamille, 2020-09-09:
Can you explain what optimizing data blocks really does on the Plc?
Is it just rearranging things, so that serialization gets more complicated?
Can you still read the whole DB as once byte-array, assuming you know the optimized total size?

The problem, besides the effort to implement would definitely be figuring out how Siemens does this DB optimization. If there is some well-defined algorithm it would be still quite a huge undertaking; if not it also requires a ton of reverse-engineering.

And if optimized means you can no longer just read a raw amount of bytes from the data block, then there is probably no solution for it at all.

FalcoGoodbody, 2020-09-09:
Optimized Datablocks are very good evolution in the S7 plc world.
When using optimized data blocks the memory usage of the data block is managed by the system. There are no spaces between the variables and one can create structs in a way they want.

A good explanation gives the programming guideline from SIEMENS which can be found in englisch language [here](https://cache.industry.siemens.com/dl/files/040/90885040/att_970576/v1/81318674_Programming_guideline_DOC_v16_en.pdf) and is available in other languages too.

The system rearanges the variable inside the data block from big type like dword or real at the top to the smaller one like int and word until bool so that all information is in one line and there are no spaces anymore between.

mycroes, 2020-09-09:
There's a document (by Siemens) about optimized vs non-optimized datablocks. There's a few things I remember (about optimized datablocks):

- Data is (re-)ordered by type, I believe larger variables first and then smaller to the end, all bits joined together so space is saved.
- The order of bytes wthin variables is different, because the endiannes of the PLC differs from previous series. This might perhaps be the main reason why this is called optimized and what we need for S7NetPlus is non-optimized; the PLC can't do any calculations on these items without converting the data to the correct endiannes.

I believe [this Wireshark dissector](https://sourceforge.net/projects/s7commwireshark/) has come a long way to parsing the S7 protocol that's used for optimized datablocks. I have no idea whether this is used by more than just TIA portal and if there are other libraries that already implement this protocol. I do know it looks a lot more complicated than what we do now 😛.

FalcoGoodbody, 2020-09-09:
yes it is more complicated when you want to read to a PC program. For the plc-programmers it is much more easier now.
But anyway it is a special thing by TIA. A lot of other software already support this like some OPC-Server software (for example TANI, INSEVIS, and so on). 

And you are correct. They re-ordered the data by type from large to small.

As said this is maybe something for the far future :-)

scamille, 2020-09-09:
Well that does not sound too complicated in theory. Ordering by Size of VarType is quite easy (I love LINQ), and converting with LittleEndian is a lot easier since one can just use System.BitConverter which is nearly always little Endian as well.

@FalcoGoodbody: Can you test the following:
Create a optimized DB with Byte[100] in it, and then see what happens when you peform Plc.ReadBytes and Plc.WriteBytes using S7.NetPlus? Can you write and read all chunks in the array?

At least OPC UA has a completely separate access model by variable name/path only, so they really do not care about addresses at all. And yes you also get little endian from a S7 1500 with it. (which was quite annoying, had to massively adjust our conversion library at work).

FalcoGoodbody, 2020-09-09:
For sure I can. Will do so tomorrow when back at work.

What do you mean with "write and read all chunks..." ? Sorry I do not know this word

scamille, 2020-09-09:
I just meant if you can read/write all entries at all indexes throughout the array. Couldn't come up with a better word for the sections which together build an array.
https://en.wikipedia.org/wiki/Array_data_structure says: ```an array, is a data structure consisting of a collection of elements```

FalcoGoodbody, 2020-09-09:
Thank you.
Array is clear for me...chunks not but it is ;-)
Will test for you tomorrow

mycroes, 2020-09-09:
I don't think you can access optimized datablocks using the current 'methods' of the S7 protocol. Also, ordering is only possible if you know all defined fields in the datablock, so you can't read a single variable without knowing all the other variables. I guess that's also the reason that TIA portal doesn't show offsets anymore if the datablock is optimized 😉

@FalcoGoodbody is the source for that library (from german sps-forum) available? I don't mind reading German, but if you can point me to it that's way easier 😎.

FalcoGoodbody, 2020-09-09:
yes it is available...even on Github

https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary

mycroes, 2020-09-09:
@FalcoGoodbody could you by any chance test if you can read an optimized datablock from an S7-1500 using DotNetSiemensPlcToolbox? If that works I'll make sure to add it to S7NetPlus as well...

FalcoGoodbody, 2020-09-10:
For sure I will.
Maybe on today or on Friday.

FalcoGoodbody, 2020-09-10:
@scamille I just tried to read 100 bytes from the optimized data block an I imediately get an exception

`
S7.Net.PlcException
  HResult=0x80131500
  Nachricht = Invalid response from PLC: '32-03-00-00-00-00-00-02-00-04-00-00-04-01-05-00-00-00'.
  Quelle = S7.Net
  Stapelüberwachung:
   bei S7.Net.Plc.ReadBytesWithSingleRequest(DataType dataType, Int32 db, Int32 startByteAdr, Byte[] buffer, Int32 offset, Int32 count)
   bei S7.Net.Plc.ReadBytes(DataType dataType, Int32 db, Int32 startByteAdr, Int32 count)
   bei S7NetPlus_Testproject.Program.Main(String[] args) in C:\Users\Robert.Oppel.OPL-NB-008\source\repos\S7NetPlus-Testproject\S7NetPlus-Testproject\Program.cs: Zeile34

Innere Ausnahme 1:
PlcException: Invalid response from PLC: '32-03-00-00-00-00-00-02-00-04-00-00-04-01-05-00-00-00'.
`

FalcoGoodbody, 2020-09-10:
@mycroes I am sorry but I do not understand how I can use this library (DoNetToolbox) . There is nearly no docu (or I can not find it) and when I import the whole github code, I get messages that some projects could net loard correctly and I can not compile or use one of the projects included in the solution map. I do not know how to solve this or what I am doing wrong

mycroes, 2020-09-10:
@FalcoGoodbody OK, thanks anyway. The library is a bit of a mess (as is S7NetPlus as well, to be honest, but we're working on that), so I'm not sure either. I'll see if I can write something that should work, then maybe you can test it. I might be able to test this myself as well, but I don't even have TIA portal installed so I'd still need to rely on someone else to prepare the PLC.

FalcoGoodbody, 2020-09-10:
Well as I mentioned before this should not be a high priority point to work on. Lets remember it for the far future.

stefanak-michal, 2025-05-30:
https://github.com/S7NetPlus/s7netplus/issues/210