S7NetPlus/s7netplus #237: How to handle ErrorCode correctly?

FalcoGoodbody, 2019-10-08:
Dear users,

I would like to connect to 2 PLCs (S7-300) wich are connected behind 1 router which has port-forwading implementet.

Router-IP: 192.168.0.80
SPS1-IP: 192.168.2.62
SPS2-IP: 192.168.2.63

Port-Forwading as followes:
192.168.0.80:102 => 192.168.2.62:102
192.168.0.80:103 => 192.168.2.63:102

Unfortunately the connection cannot be established.
I would like to check for detailed errors and therefore I want to use "ErrorCode" how it is descriped in the documentation.
But I do not know how to. The Plc.Open()-Method does not return an ErrorCode.

Can some give me a hint how to implement it correctly please?

Thank you in advance!

allatrue, 2019-10-08:
Try to check first the connectivity with windows telnet command:
Telnet 192.168.0.80 102

Maybe the router is not configured properly.

FalcoGoodbody, 2019-10-08:
Well thank you.
It seems as if the router isn´t configured correctly.
The connection can not be established.

But how can the ErrorCode be handled if the Plc.Open-method does return only void?

I tried it this way

`            try
            {
                plcRNV1.Open();
            }
            catch (Exception)
            {
                lastErrorCode.GetTypeCode();
                MessageBox.Show(lastException.Message);
            }`

The message is, that the connection couldn´t be established with ErrorCode "no Error".
That does not make any sense to me.

FalcoGoodbody, 2019-10-08:
Ok problem solved...the router was not configured correctly.

closed 2019-10-08 by FalcoGoodbody