S7NetPlus/s7netplus #238: Asynchronous Communication

Pacheko, 2019-10-09:
I am trying to read 1665 tags asynchronously, but I am getting incompatible / invalid TPKT failure.

Synchronously the performance is in 3 seconds, needed to be faster.

I am going through a list and returning to it, the code is as follows:

var value = await s7client.ReadAsync (dataType, db, startByte, varType, count, bit);

Can you tell where the error is?

FalcoGoodbody, 2020-09-09:
Hi @Pacheko,

there was an update of this library some days ago (0.7.0) which includes some major changes in the communication.
Maybe you can give it a try and retest with this version?

Pacheko, 2020-09-11:
Good Morning,

I am reading synchronously, but now I have 2300 tags.
I am in 600 to 800 ms of reading.
I will not say it is good, but for so many tags it is acceptable.

The solution at the time to improve this was to read the dbs as follows:

  listDataItem.Add (new DataItem () {DataType = dataType, VarType = varType, DB = db, BitAdr = bit, Count = count, StartByteAdr = startByte, Value = new Object ()});

s7client.ReadMultipleVars (listDataItem);

FalcoGoodbody, 2020-09-14:
Thanks for that hint.
I tried to read multible vars and it works fine.
Even reading StringEx (S7-String @scamille ) worked really well.

I then measured the time it took to read 7 variables, each another datatype, synchronously with the standard method and converting them. After that I did read the same 7 variable with "ReadMultipleVars" and it was soooo much faster. Wow.

With standard method and it takes between 150 - 270 ms.
With ReadMultipleVars it takes something between 23 - 28 ms.

scamille, 2020-09-14:
Well reading variables separately has a very high network overhead and multiple round-trips for each call, so it is expected to get a high time there.

ReadMultipleVars tries to read multiple variables in a single request. Those variables can even be in different DB's and whatnot. But there are some low restrictions on the maximum amount of variables and total size you can read (basically nothing about max PDU size).

If you have many variables in a single DB in some continuous memory range, I would recommend using the ReadClass or ReadStruct functions, or ReadBytes and manually performing the desired serialization (but there currently is no good support for that in S7.NetPlus as far as I can tell).

Pacheko, 2020-09-14:
Does anyone know if I can communicate with logo8! with this library?

FalcoGoodbody, 2020-09-15:
I am not really sure. There is an other issue #179 where this already discussed but it seems as if it is not possible.

FalcoGoodbody, 2020-09-15:
@scamille is there a method "WriteMultipleVars"? Could not finde one.

scamille, 2020-09-15:
```public async Task WriteAsync(params DataItem[] dataItems)``` seems to be able to write a list of DataItems.

mycroes, 2020-09-15:
> Does anyone know if I can communicate with logo8! with this library?

Please test, I added the required ID's for logo8 which should permit connecting. I'm closing this issue though (because it's fixed), feel free to continue the discussion in #179 with regards to logo8, or open a new issue.

closed 2020-09-15 by mycroes