S7NetPlus/s7netplus #242: Is it possible to implement a DB Scanner with this library?

masoudium, 2019-12-11:
The S7netplus is a great library for sure. But the way I'm using it is by read("DB Address") function which needs the DB address to access the data. Assuming I need to find out the DBs on a plc without having the PLC source project. I find out that read function throws an exception if the presented address is not valid in PLC. But I want a way to find the data type I mean the byte, word or dbword...
Can anyone help me with this problem?

masoudium, 2019-12-23:
NO comment?

mycroes, 2019-12-23:
Sorry for the late reply, unfortunately I'm pretty busy with other stuff. It's not possible to scan the datatype because it's all stored as blocks of bytes without any header. Possibly there's metadata somewhere in the program blocks, but it's not something I desire to explore.

Closing this issue as I don't see this as something that can easil be added to S7NetPlus.

closed 2019-12-23 by mycroes