S7NetPlus/s7netplus #243: S7 v 0.4.0  has no error processing

dimafainberg, 2019-12-15:
in v 0.4.0  has no error processing(var result = plc.Open();),      plc. Open return void  
thanks  cant be equal to var.. can you help PLZ thnaks

timverwaal, 2019-12-16:
I had the same issue. The solution however is simple. 
The `Open()` method throws an exception when something goes wrong.

```
using (var plc = new Plc(CpuType.S71500, "192.168.0.1", 0, 1))
{
    try
    {
        plc.Open();

    } catch (Exception e)
    {
        // This code is only executed when something went wrong with Open();
        Console.WriteLine(e.Message);
    }           	
}
```

dimafainberg, 2019-12-17:
Thank you !! But if a problem with read ? in previous version was a LastErrorCode

timverwaal, 2019-12-17:
This works the same. I looked up the commit for this change: https://github.com/S7NetPlus/s7netplus/commit/1538de148b4aa48f845f14181cfa4ad5cd7ccbf2 (_"Replace LastErrorCode and LastErrorString with exceptions"_)

dimafainberg, 2019-12-17:
Thanks now i see !!!!

mycroes, 2020-09-04:
Closing, feel free to reopen if you're still running into problems with how this changed (`LastErrorCode` vs Exceptions).

closed 2020-09-04 by mycroes