S7NetPlus/s7netplus #248: S7-1212C Can't read data.

WontheMikeLi, 2019-12-23:
The exception is "Unable to read data from the transport connection: An existing connection was forcibly closed by the remote host."

Version 3.0 also has the same exception.
Sample code is below

            try
            {
                if (objTemp_Plc.IsConnected)
                {
                    //byte[] readByte = objTemp_Plc.ReadBytes(DataType.DataBlock, 1, 0, 1);
                    object objTemp = objTemp_Plc.Read(DataType.DataBlock, 1, 0, VarType.Byte, 1);
                    //object uscita = objTemp_Plc.Read("DB1.DBB0");
                }
            }
            catch (Exception ex)
            {
                throw ex;
            }

Sharp 7 can read data.

mycroes, 2019-12-23:
Can you try without plc.IsConnected? I believe that's still bugged (i.e., destroying the connection).

WontheMikeLi, 2019-12-24:
Thanks a lot, as what you said, after comment plc.IsConnected it works like a charm.

closed 2019-12-24 by WontheMikeLi