S7NetPlus/s7netplus #249: WriteBytes not working

nayanavv, 2020-01-03:
Writing bits to the WriteBytes method not working. I'm using the following code to write the bits, but the SetBit/ClearBit function is not giving any result in the PLC

   bool control;
   byte[]  db1Bytes = new byte[1];
   if (control == true)
   {
      S7.Net.Types.Boolean.SetBit(db1Bytes[0], 0);
   }
   else
   {
       S7.Net.Types.Boolean.ClearBit(db1Bytes[0], 0);
   }
   plc.WriteBytes(DataType.DataBlock, 1, 0, db1Bytes);

Can anyone tell me how to solve this issue?
plc.Write("DB1.DBX0.0", true); works fine!

FalcoGoodbody, 2020-01-06:
Hello @nayanavv ,

I use this method to write a bytearray/string into a PLC and it works fine so fare.

`plcKonti2.WriteBytes(DataType.DataBlock, 156, 0, bytearray);`

dattatrayshindebuck, 2021-05-07:
Hello @FalcoGoodbody,

I am facing the same issue as @nayanavv is facing. 

I am using multithreading to write and read bytes and that has nothing to do with functions working properly.

I would like to Answer your previously asked questions.
The adresses in your PLC are correct? The DB1 exists? - **Yes**
Does the method "WriteBit" works? - **No. I tried to debug my program. The step of writing bits is executed. I checked my array right before the breakpoint and I found that the bit was not set** 
Do you get any error message? - **No. There is no error message and unfortunately my thread is also automatically killed if I try to write the bytes.** 

@mesta1  - I am following your video on Youtube.

Thanks in advance.

Regards.

mycroes, 2021-05-07:
> I am using multithreading to write and read bytes and that has nothing to do with functions working properly.

Not to rain on your parade, but S7NetPlus is not (currently) thread-safe. You could very well be right about the `WriteBit` method though. Can you share the code you're using? I'm guessing it's a minor issue. I'm a bit short on time for the larger tasks at the moment, but I expect I could fix this this evening if I have a sample. Might actually be related to the original issue at hand (`SetBit` perhaps not working), but I'm assuming your code differs.

dattatrayshindebuck, 2021-05-07:
using S7.Net;
using System.Linq;
using System.Threading;
using System.Windows;

namespace WPFSiemens
{
    public class PCtoPLC 
    {
        public bool IsConnected = false;
        public bool Kill ;
        Plc SiemensPLC = new Plc(CpuType.S71200, "192.168.8.254", 0, 1);
        WPFSiemens2.DB_ausgange db_Ausgange = new WPFSiemens2.DB_ausgange() ;
        WPFSiemens2.DB_Eingaenge dB_Eingaenge = new WPFSiemens2.DB_Eingaenge();

        public void Connect()
        {
            System.DateTime ConnectionTimeout = System.DateTime.Now; ;
            System.DateTime now = System.DateTime.Now;

            if (IsConnected == false)
            {
                MessageBox.Show("Connecting to" + "192.168.8.254", "Title");
                IsConnected = true;
                ConnectionTimeout = System.DateTime.Now;
                SiemensPLC.Open();
                MessageBox.Show("Connected to" + "192.168.8.254", "Title");

                if (SiemensPLC.IsConnected)
                {
                    StartCommunication();
                }
            }
            else
            {
                MessageBox.Show("Already Connected", "Title");
            }
        }

        public void StartCommunication()
        {
            ThreadStart threadStartA = new ThreadStart(UpdateAusgange);
            Thread updateAusgange = new Thread(threadStartA);

            updateAusgange.Start();
        }

        private void UpdateAusgange()
        {
            do
            {
                var DBAusgaenge = SiemensPLC.ReadBytes(DataType.DataBlock, 100, 0, 133);
              .
              .
              .
              .// works perfectly fine

           dB_Eingaenge.E_230v_s1 = true;
            do
            {
                byte[] DBEngangeNew = new byte[604];
               
                {
                    if (dB_Eingaenge.E_230v_s1 == true)
                    {
                        S7.Net.Types.Boolean.SetBit(DBEngangeNew[1], 4);
                    }

                    SiemensPLC.WriteBytes(DataType.DataBlock, 101, 0, DBEngangeNew);
               
                } while (Kill == false);
            } while (Kill == false); 
         }

       

I am new to programming and suggestion are welcome.

dattatrayshindebuck, 2021-05-07:
Many thanks for the support. Below is my code snippet.

using S7.Net;
using System.Linq;
using System.Threading;
using System.Windows;

namespace WPFSiemens
{
public class PCtoPLC
{
public bool IsConnected = false;
public bool Kill ;
Plc SiemensPLC = new Plc(CpuType.S71200, "192.168.8.254", 0, 1);
WPFSiemens2.DB_ausgange db_Ausgange = new WPFSiemens2.DB_ausgange() ;
WPFSiemens2.DB_Eingaenge dB_Eingaenge = new WPFSiemens2.DB_Eingaenge();

    public void Connect()
    {
        System.DateTime ConnectionTimeout = System.DateTime.Now; ;
        System.DateTime now = System.DateTime.Now;

        if (IsConnected == false)
        {
            MessageBox.Show("Connecting to" + "192.168.8.254", "Title");
            IsConnected = true;
            ConnectionTimeout = System.DateTime.Now;
            SiemensPLC.Open();
            MessageBox.Show("Connected to" + "192.168.8.254", "Title");

            if (SiemensPLC.IsConnected)
            {
                StartCommunication();
            }
        }
        else
        {
            MessageBox.Show("Already Connected", "Title");
        }
    }

    public void StartCommunication()
    {
        ThreadStart threadStartA = new ThreadStart(UpdateAusgange);
        Thread updateAusgange = new Thread(threadStartA);

        updateAusgange.Start();
    }

    private void UpdateAusgange()
    {
        do
        {
            var DBAusgaenge = SiemensPLC.ReadBytes(DataType.DataBlock, 100, 0, 133);
          .
          .
          .
          .// works perfectly fine

       dB_Eingaenge.E_230v_s1 = true;
      
            byte[] DBEngangeNew = new byte[604];
           
            {
                if (dB_Eingaenge.E_230v_s1 == true)
                {
                    S7.Net.Types.Boolean.SetBit(DBEngangeNew[1], 4);
                }

                SiemensPLC.WriteBytes(DataType.DataBlock, 101, 0, DBEngangeNew);
           
         
        } while (Kill == false); 
     }
I am new to programming and suggestion are welcome.

FalcoGoodbody, 2021-05-07:
Well I use "WriteBit" with success and "WriteBytes" as shown in the 1st example and as follows:

```csharp
            // Auftragsnummer in byte-Array wandeln                   
            byte[] byteArray;

            if (CONDITION)
            {
                byteArray = ConvertStringToByteArray(STRING);
                
                try
                {
                    if (plc.IsConnected)
                    {
                        plc.WriteBytes(DataType.DataBlock, 156, 0, byteArray);
                    }
                }
                catch (Exception)
                {
                    throw;
                }
            }
```

I do not use "WriteBytes" any often because I mainly read information from the PLCs.

dattatrayshindebuck, 2021-05-10:
@FalcoGoodbody  Thank you for extended replies.

WPFSiemens2.DB_Eingaenge is a C# class structure I have used to create variables. I started with the idea of writing the whole class to the PLC but I did not find that method intelligent. One has to think a lot before making class with addresses very similar to the PLC database.

commit 0b8bd66b (2021-05-10): Fix bitwise operator in Boolean.ClearBit

Fixes #249
closed 2021-05-10 by mycroes