S7NetPlus/s7netplus #250: "TPKTInvalidException: TPKT is incomplete / invalid" occur when Read and Write

jun-l-lin, 2020-01-05:
`
bool isWrite=false;
public void BoolWrite(string add, bool val)
{
     isWrite=true;
     plc.Write(add,val);
     isWrite=false;
}

public void ClassValRead(Variable var, int db)
{
     if (!isWrite)
     {
               plc.ReadClass(var,db);
     }
.......
}

`

a Thread cycle read "Variable" class, when I write a single variable to plc, the issue occur:

    TPKTInvalidException: TPKT is incomplete / invalid
    S7.Net.TPKT.Read(System.IO.Stream)
    S7.Net.COTP.TPDU.Read(System.IO.Stream)
    S7.Net.COTP.TSDU.Read(System.IO.Stream)
    S7.Net.Plc.WriteBitWithASingleRequest(S7.Net.DataType, int, int, int, bool)

MikeK-10, 2020-04-14:
I am having a similar issue, based on the very simple sample code at: https://www.fontenay-ronan.fr/siemens-plc-and-c-communication-simatic-step7plcsim-visualstudio/

The timer will successfully run and read values from the PLC for a short period of time (up to a minute or two), I then get the TPKT error and subsequent commands fail :( 

I have tried to worked around by creating a new instance of the plc but this gives the same result - I have to close and reopen the whole project :( 

I am using v0.4.0, .NET Framework 4.8 and visual studio 2019

mycroes, 2020-04-15:
Hi guys,

Sorry for my late response to issues. This specific exception is quite useless, while the PLC actually provides some sort of error code normally. I'll see if I can add this sometime soon, since it's one of the most recurring issues. @MikeK-10 since it does work for you first off and it stops working later, this definitely seems to be an issue with the plcsim and not S7NetPlus/Sally7 (bonus points for testing that as well). If you want to discuss that further please open a separate issue.

ccaiwl, 2020-07-08:
遇到同样问题：  S7.NET.PlcException:"TPKT is incomplete/invalid"    内部异常 "TPKTInvalidException: TPKT is incomplete / invalid"

mycroes, 2020-09-04:
Please retest against 0.7.0, possibly fixed or providing a more useful exception.

mycroes, 2021-06-04:
Closing, please reopen if the issue persists with S7NetPlus 0.12.0.

closed 2021-06-04 by mycroes