S7NetPlus/s7netplus #251: Read IW

sudak-91, 2020-01-19:
Hello. How i can read analog input?

Mallowan, 2020-01-20:
Hello,
the handling with analog plc inputs supposes to have respective analog modules. Do you agree?
Then the having of onboard analog modules provides an opportunity to digitize your analog signal.
Correspondingly the digitized signal you can pass whenever you want nowadays. Even with this library.

hope this helps.

sudak-91, 2020-01-20:
Can i use PLC.Read("IW20")?

Mallowan, 2020-01-20:
first, you have to pass that input word to the word which is located in Global DB, then read it from there.

sudak-91, 2020-01-20:
I can read this signal from DB, because i forgot switch off  db
optimisation.

FalcoGoodbody, 2020-01-20:
Can you turn the optimization off? You just need to do this for the DB you want to read from.

mycroes, 2020-09-04:
@sudak-91 I think you can now read IW20, can you test this on 0.7.0?

mycroes, 2021-06-06:
Should've been fixed for a while now, closing. Feel free to reopen if there's still issues reading IW.

closed 2021-06-06 by mycroes