S7NetPlus/s7netplus #252: Unable to read PLC DB data

05749, 2020-01-29:
Hi, I am using the S7-1200 and TIA portal V1.4.
I am running Readbytes from the S7.net library, but I get the following error:
 "Unable to read data from the transport connection: An existing connection was forcibly closed by the remote host."
I can do plc.open () and plc.close () without any problems.
Is there any solution? Any advice is welcome.
thank you in advance.
[image]

FalcoGoodbody, 2020-01-31:
Did you check that the DB is not optimized?

05749, 2020-01-31:
Thank you for your comment.
Yes, I unchecked DB optimization according to the documentation.
[image]
Do I need other settings on the PLC besides the contents of this document?
Thank you.

allatrue, 2020-02-01:
Did you enabled PUT/GET communication for the CPU?

05749, 2020-02-01:
Yes, I have enabled PUT / GET communication for the CPU .

mycroes, 2020-02-02:
Try to remove the Plc.Isconnected check, it's broken.

05749, 2020-02-02:
I thought Plc.Isconnected was a way to verify communication with the PLC, but does it affect Plc.ReadBytes?

mycroes, 2020-02-02:
I believe it accidentally terminates the connection. I'm way behind on fixing issues like this, because obviously this shouldn't be the case. Let me know if it works without the IsConnected check.

05749, 2020-02-03:
Thank you for your comment.
I was able to read the DB by removing the IsConnected check.
[image]
thank you very much.

FalcoGoodbody, 2020-02-04:
Well...that´s strange.
I am using the PLC.IsConnected to read data from a S7-1500 at there it works fine so far.
Maybe it is problem depending to the PLC used?

mycroes, 2020-02-04:
Maybe one of you isn't using the latest version?

FalcoGoodbody, 2020-02-04:
I´m using Version 0.4.0 from the last year. Via NuGet there is no newer one available.
For me everything works fine and I am absolutely happy with this library.

05749, 2020-02-05:
I using the same S7.Net version 0.4.0.
However, the PLC I'm using is S7-1200 and the CPU firmware version is V4.1.
Do these conditions affect anything?

mycroes, 2020-09-08:
0.7.0 has some major improvements and better exception message on errors, so please retest.

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes