S7NetPlus/s7netplus #253: read/write in OB1

AthanasiosT, 2020-02-10:
I would like to create a simple HMI for a s7-1200.
Is it possible to read write to OB1?
I need to change memories M0.0 and M0.1, read inputs I0.2, I0.3, I0.4, I0.5  and outputs Q0.0, Q0.1, Q0.2, Q0.3.
Do I need to write the Ladder code in a FC and then call the FC in the OB1?
How should I syntax the read and write commants?
Thanks in advance.

Mallowan, 2020-02-11:
There's document to this library. True, it is outdated. 
However, it is still showing main abilities of library.

AthanasiosT, 2020-02-11:
The only thing I read in the documentation is that you can read and write only in DBs. Even in you tube videos they read and write from DBs.
What if you want to read and write booleans directly from OB1, on a simple ladder program?
You can't uncheck Optimized block access to the OB1.
Will it work?

FalcoGoodbody, 2020-02-20:
Hello,

I recommend to read and write only to DBs. That is the best practice.
You could read/write to Instance-DBs of FB but I do not recommend this. I case of an fault in adressing the variables you directly affect the program logic of the PLC program which can cause big problems.

When a PLC programmer creates a HMI the variable to communicate with/from the HMI a beeing stored in DBs. I suggest do change the PLC program in a way that all signals you want to read or write are stored in a separate DB.

DeputyOfCopyPaster, 2020-02-20:
Falco, it seems you've missed my mails. Or?

Mallowan, 2020-02-20:
Falco, it seems you are missing my mails. Or?

FalcoGoodbody, 2020-02-20:
No I don´t, sorry.

I tested you Code at a PLC S7-1516F and sent you a screenshot, didn´t I?

Was there something more you wanted to know?

mycroes, 2020-09-04:
@AthanasiosT your original question was whether you can read/write OB1, I think that should work since one of the past releases. I'm closing this because I expect it's fixed and @FalcoGoodbody gave a good alternate solution, but feel free to reopen if you need more help.

closed 2020-09-04 by mycroes