S7NetPlus/s7netplus #255: How do I access to Plc (instance) from other classes.

Mallowan, 2020-02-11:
hello,
for example, there's known way to do that in MainWindow.
`
public Plc plc;
plc = new Plc(CpuType.S71200, "100.1.1.10", 0, 0);

`
Could anyone share an example, which shows access to Plc from other classes?

thank you in advance!

MiroslavMikus, 2020-05-04:
Hey, the base idea should be to use [Dependency Injection](https://en.wikipedia.org/wiki/Dependency_injection).

We are using Autofac. Here is the simplified version:
```
var builder = new ContainerBuilder();

builder.Register<Plc>(()=> new Plc(CpuType.S71200, "100.1.1.10", 0, 0));

var scope = builder.Build();

var yourService =  scope.Resolve<Service>();
```

Where the `Service` requests the `Plc` in the constructor:

```
public class Service
{
    public Service(Plc myPlc)
    {

    }
}
```
Let me know if you have still some problems :)

Wobeam, 2020-05-04:
Hi there,
I'll try to replicate, then there will be questions definetely.
Thanks!

Mallowan, 2020-05-04:
hi there,
I will try to replicate, then there will be questions definetely.
Thanks!

OOOOO1, 2020-05-30:
[image]

OOOOO1, 2020-05-30:
> 嘿，基本思路应该是使用[依赖注入](https://en.wikipedia.org/wiki/Dependency_injection)。
> 
> 我们正在使用Autofac。这是简化版：
> 
> ```
> var builder = new ContainerBuilder();
> 
> builder.Register<Plc>(()=> new Plc(CpuType.S71200, "100.1.1.10", 0, 0));
> 
> var scope = builder.Build();
> 
> var yourService =  scope.Resolve<Service>();
> ```
> 
> 当`Service`请求`Plc`在构造函数：
> 
> ```
> public class Service
> {
>     public Service(Plc myPlc)
>     {
> 
>     }
> }
> ```
> 
> 让我知道您是否还有一些问题:)

您好，按照你说的，我尝试复制你的代码，结果出现
Error	CS1593	Delegate 'Func<IComponentContext, Plc>' does not take 0 arguments

MiroslavMikus, 2020-05-30:
Hey, I do speak just ENG / GER so maybe this answer will not be directly on point. Anyway, it looks like you are using a different version of Autofac. Pls try this:
``` csharp
builder.Register<Plc>(CONTEXT => new Plc(CpuType.S71200, "100.1.1.10", 0, 0));
```
A second fix could be to bump your Autofac dependency to at least 4.9.1

Best regards Miro