S7NetPlus/s7netplus #256: Problems reading and writing the input (IB) and output(QB) data area

DimalRus, 2020-02-14:
Hello
I using PLCSIM, read the output data(QB 0),
ReadDataFromPLC = plc.readbytes(DataType.Output, 1, 0, 1);

At the same time, when using the controller in hardware, I do not use this data area.
In the controllers 300 and 400, PLSSIM version 5.4.8 is launched, respectively, and for 1200 and 1500 version 16.
Reading in the 1200 and 1500 controllers works, and 300, 400 does not work.
It is imperative to read down the output and input areas.

an error occurs while reading WrongNumberReceivedBytes

mycroes, 2020-02-14:
I don't know about any differences with regards to outputs between 300/400 and 1200/1500. Not entirely sure though, because I've never used output areas myself and I also never used 300/400 series PLC's. Have you ever tried another open source library to see if that does work for both PLC types?

DimalRus, 2020-02-14:
I used S7PROSIMLib.dll. It was used only for PLCSIM v5.4; it successfully read the output and recorded the input.

DimalRus, 2020-02-14:
When using 1200 and 1500 PLC, the addresses of the inputs that I want to write cannot be used in the hardware configuration. If this condition is met, then the write is successful and the reading is the same.

DimalRus, 2020-02-14:
But this additional condition when using PLCSIM v5.4 and PLC series 300 or 400 does not help, the error WrongNumberReceivedBytes occurs

mycroes, 2020-09-04:
Can you retest this against 0.7.0? Several issues that might be related to your problem have been fixed, so I expect this issue might be fixed now as well.

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes