S7NetPlus/s7netplus #257: Cant read or write in 0.4.0 s71200 tia portal v14

yusuflars, 2020-03-02:
![Screenshot_1](https://user-images.githubusercontent.com/61707883/75705232-09757d80-5ccc-
[image]
[image]

11ea-9b26-2e14e0685cad.png)
[image]
[image]

[image]

Always getting "IndexOutOfRangeException"

zoltraks, 2020-03-08:
This might be cause by not having "Protection & Security" option "Permit access with PUT/GET communication from remote partner" checked.

I have also put a Pull Request with small fix for that:

https://github.com/S7NetPlus/s7netplus/pull/258

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes