S7NetPlus/s7netplus #259: Reading class / struct with gaps in the DB

Equino3108, 2020-03-09:
Hi there,

How would I use ReadClass if my DB looks like this:

DB100.DBX0.0 BIT
DB100.DBX0.1 BIT
DB100.DBX0.2 BIT
DB100.DBX1.0 BIT
DB100.DBX1.1 BIT
DB100.DBX1.2 BIT

I understand if the next Byte is of a different type than BIT, it works because of the Math.Ceil numbytes logic in the code, but how would S7.net know there is a gap of 5 bits if the next property is bool again?

Looking at the implementation of GetIncreasedNumberOfBytes(), I'm pretty sure this wouldn't work:

public bool Bool0_1 {get; set;}
public bool Bool0_2 {get; set;}
public bool Bool0_3 {get; set;}
public bool Bool1_1 {get; set;}
public bool Bool1_2 {get; set;}
public bool Bool1_3 {get; set;}

Do I have to declare the properties with type byte in this case? And then extract the single bits on my own? Or is there a better solution?

Thanks

FalcoGoodbody, 2020-05-14:
> Do I have to declare the properties with type byte in this case? And then extract the single bits on my own? Or is there a better solution?

I would recommend to do it this way. Declare with type byte and handle it within the class

mycroes, 2021-06-04:
I would recommend to use separate addressing instead. Closing this issue, feel free to re-open if more questions arise.

closed 2021-06-04 by mycroes