S7NetPlus/s7netplus #262: TKPT error after initial connection success

MikeK-10, 2020-04-15:
Addition to issue discussed #250 and new issue as requested. 

I am not using plcsim - I have an actual S7-1200 connected

ccaiwl, 2020-07-08:
遇到同样问题  不定时的出现异常   真机s7-200 smart

mycroes, 2020-09-04:
Hi @MikeK-10, can you retest against 0.7.0?

MikeK-10, 2020-09-15:
Hi, still get the failures, error message now is:
"Couldn't establish the connection to 192.168.1.20. 
Message: TPKT header is incomplete / invalid. Received Bytes: 0 expected:4"

Restarting the project allows successful re-connection for a short period (30s to 1min)

mycroes, 2021-06-04:
Hi @MikeK-10, perhaps it's a good idea to retry with 0.12.0. I've now implemented synchronization in the library, which is something that was really missing for a lot of people. Not sure if your issue relates to that, but if it's still not fixed then I'll remove the retest label and we can see what we can try next.

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes