S7NetPlus/s7netplus #264: The relationship with PROFINET

mytion7, 2020-05-08:
I tried to communicate with siemens plc through PROFINET.

I found this library, but I'm not sure this library support PROFINET or not.

If I use this library, can I says that my program communicates with PLC through PROFINET?

RoyHeerkens, 2020-05-11:
The library uses s7 protocol, this is a different protocol than Profinet.

mycroes, 2020-05-12:
@RoyHeerkens thanks for answering, to be honest I don't have much PLC experience, so I'm not sure about the internals of the Profinet protocol. S7NetPlus does indeed use the S7 protocol, which seems to be a proprietary Siemens protocol with almost no documentation.

mycroes, 2021-06-04:
Let's answer this properly now. S7 and Profinet are entirely different. There are libraries available for Profinet communication and there are vendors that sell dedicated hardware for Profinet. Perhaps the latter also support the S7 protocol in some cases, but the protocols don't share any technical details.

closed 2021-06-04 by mycroes