S7NetPlus/s7netplus #265: Why there's so huge difference between usage in usual class and static class?

Wobeam, 2020-05-18:
Hello,
could anyone answer, why when I create instance of Plc in usual class the library functions pretty fine and I can read and write vars. At least it functions correctly.
However, when I tried a variant as shown below. I had gotten incorrect function of library.

**an usage in Static class:**

```
 public static class Connection
    {
        public static Plc plc { get; }
        public static bool IsConnected { get; }
        public static bool IsAvailable { get; }
        public static ErrorCode errorCode { get; }
       
        static Connection()
        {
            try
            {
                plc = new Plc(CpuType.S71500, "192.168.1.1", 0, 0); 
                if (!(IsAvailable = plc.IsAvailable)) throw new Exception("controller not responding");
                errorCode = plc.Open();              
                if (errorCode != ErrorCode.NoError) throw new Exception(plc.LastErrorString);
                if (!(IsConnected = plc.IsConnected)) throw new Exception("connection failed");
            }
            catch (Exception ex)
            {                             
                var messageBoxResult = MessageBox.Show
                ("Attention!", ex.Message, MessageBoxButton.OK, MessageBoxImage.Error);
            }
        }
    }

```

**Access to plc methods:**
`Connection.plc.Write("DB10.DBX2.3", 1); `

mycroes, 2020-05-19:
Because the operations are not thread-safe. You're probably invoking new actions on the PLC before running actions have completed, which is not supported / not possible. Some PLC's do support concurrent requests (i.e. the 1500 supports 3), but it would require quite some changes to the S7NetPlus internals.

If you can share some details about your project I can tell you some ways to implement synchronized access to the PLC.

Wobeam, 2020-05-19:
@mycroes, first of all thanks for your input.
Those phenomenons I described, they are simple my assumptions. Honestly speaking I even don't know what is exactly going on when I use a static class_. In other words glitch goes by glitch.
And by the same plc logic and by implemented none-class everything looks good. And logic in Model is exactly th same. That confuses me.
A reason why I'm trying to use the static class is it allows to make access to Plc methods more easily. In contrast with none-static class it doesn't multiply instances of the Plc object (it is supposed structured project).
I would be appereciated if you take a private look on my tests. Where may I upload them?

Thanks.

mycroes, 2020-05-19:
Anywhere I can access them is fine. No promises on response time though.

Wobeam, 2020-05-19:
@mycroes, here is a link [https://mega.nz/folder/275BwTpL#d0QdX0Gsuw7nEYEo1qsd5g](url) to test. Also please read readme file in folder.

mycroes, 2020-05-20:
Hi @Wobeam,

I looked into your code and the issue probably is the multithreaded access to the PLC from both the UI and the timers in the Service. I said I could provide you with some ways to implement synchronized access, the most obvious one in your application would be to invoke all PLC calls on the UI thread. That said, that definitely isn't pretty. You're already blocking the UI thread in `Group_start()` and `Group_stop()`, so if you're having connection issues your application will become unresponsive. I'd say that if you're just developing this for your own use you could probably live with that, but if you intend to let others use this or even sell this application, then I'd say you have a long road ahead.

There's a couple of things you could do to implement synchronized access to the PLC. .NET has a `BlockingCollection` class that you could use to queue and process actions, one by one. You could start a new thread to consume items using `BlockingCollection.Take()` and push actions from other threads using `BlockingCollection.Add(...)`. If you read up on these methods then you'll notice that the `Take` method will also block the thread it's running on, however in this case that'll be the thread you specifically created for it. On the other hand, this is exactly why async has been introduced in .NET. Async primitives allow non-blocking behavior while still being able to achieve the same result in the end. S7.Net also has an async API, but you're using some old version from what I saw in the project (why don't you use the one from NuGet anyway?). You could implement a task queue that would allow for implementations similar to what I described with the BlockingCollection, while not requiring any thread to block.

Either way, I guess you would benefit most from some good programming books that could explain more about how to solve problems in general. You're using ViewModels in your sample, but they don't really offer what a ViewModel could offer due to the implementation. You have a static class because that's 'easier', while you actually have a ServiceLocator NuGet package included (even though some people would argue Service Location is an anti-pattern, it's at least a pattern that would just as well solve your problem).

I'm closing this issue because it's not related to S7NetPlus, feel free to respond in this issue while it's closed though.

Regards,
Michael

closed 2020-05-20 by mycroes

Wobeam, 2020-05-20:
@mycroes, Hello,
If this kind of Apps has a connection issue (e.g. PLC not responding), that  will mean only one thing, -nothing from this App will be sent and nothing will be received. Thus the App becomes useless and it doesn't matter if it is responsive or unresponcive after connection failure.  The controlled object must be stopped anyway. Based on this principle function all WinCC projects, at least those which I have seen up to now. PC or connection has a failure -STOP, LIFE bit failure -STOP and so forth.
**As concerned blocked GUI.** I heard rumors called "freezing GUI", however, for over a year I have a relatively big App (based on principle of  'none-static class') that is currently being tested. This App continuously polls up to 100 different types of vars (values coming to TextBlocks) and has on UI many buttons which directly write bits in PLC, similar to "Group_on/off". **Yeah**, there's something that I would improve, but this "something" is not from opera "blocked/freezing GUI". I don't know why, but I 've never seen and even operators (watching on this 10 hours a day) have never seen that something (including many values on UI) had ever frozen on UI. There's noting related to that on the screen.

Pardon, but you probably haven't caught the point of my problem. The problem exactly connected with S7net usage, though. But never mind. This area is very specific so there are very few advisors on this theme, so I'm doing everything by myself. I'm used to it.

_Some points if you are still interested in:_  
-If you are going really to examine variant **none-static** with real HW. You may either "as you said, block UI thread" with buttons Group on/off or not block it at all. It doesn't matter it will never freeze UI. It will work flawlessly all the time.

-If you are going really to examine variant **static** with real HW. It won't work properly from the beginning and it doesn't matter whether you are pushing buttons or not. By the way, it also does not freeze UI.

If you are really interested in my problems related to a building of App based on known library, simple let me know I will describe that with more details.

Hope I have expressed understandable and thanks for attention!