S7NetPlus/s7netplus #266: I can read/Write DI/DO but not AI/AO...

murfie, 2020-05-27:
By using the following code, i can read an output, for an input i use (bool)plc.Read("I0.0") and this also works. So we can work on absolute addresses. 

      
 using (var plc = new Plc(CpuType.S71500, "192.168.0.199", 0,1))
            {
                plc.Open();
                bool output = (bool)plc.Read("O0.1");
                Console.WriteLine("Q0.1 " + output);
            }
            Console.ReadKey();
[image]

Writing these inputs and outputs is also possible. However I'm trying to read/write AI/AQ absolute addresses and do not want to read them from a DB, but from the absolute adresses. (e.g.: plc.Read("IW10")). Anybody has a solution for me how to read analog values?

FalcoGoodbody, 2020-05-27:
An analog input-value represents a value from 0 to 27648 and need to be scaled into a physical unit.
Within the plcs there are some standard blocks which do this in generell and are very easy to use.
I would recommend to this within in the plc and to write the so scaled value in a DB. Then read the value from there.

If not you can try to transform the analog value into an INT32. Be aware that you need to scale this value in a physical unit.

murfie, 2020-05-27:
It is really important that i can reach the analog values by their absolute address (e.g. IW10, %IW30, %QW30, ...) and not via a DB.

Wouldn't there be an option to read write analog values the same way as digital values?
e.g.: bool mybool = (bool)plc.Read("I0.0") ==> this works...
But I want to do this for IW and QW. I tried a lot of combinations already, but nothing seems to work for analog values...

FalcoGoodbody, 2020-05-28:
Did you try to read "EW0" or "ED0"? These types exist in the code of the library.

Another try would be to read a bytearray and convert it afterwards

FalcoGoodbody, 2020-05-28:
I tried it myself with an S7-1515-2 PN

`
int test = 0;
test = Convert.ToInt32(plcGlatt2.Read("EW100"));
`

I could read the value so this should work

murfie, 2020-05-29:
Thank you, this is indeed what i was looking for and it works!

I also need to read/write the analog outputs, what would be the string that i'll have to read for an analog output? Or where can i find these different "read-types" in the source code?

mycroes, 2021-06-21:
Closing, feel free to reopen if the issue persists.

closed 2021-06-21 by mycroes