S7NetPlus/s7netplus #270: plc.Read(string) function doesn

ZombieAllen, 2020-07-06:


ZombieAllen, 2020-07-06:
There's a fault in PLCAddress.cs. 

 case "M":
                            // Memory
                            dataType = DataType.Memory;
                            varType = VarType.Byte;
                            break;

the varType should be VarType.Bit.

timverwaal, 2020-07-06:
I created a pull request for this: #246. Would this commit solve your problem?

ZombieAllen, 2020-07-07:
> I created a pull request for this: #246. Would this commit solve your problem?

Hi , Thanks for your information. It works. I met a fault after compling.

Unable to locate the repository that contains the directory'E:\PersonalProject\Ref\s7netplus-PLCAdddres-Parsing\S7.Net'. S7.Net C:\Users\Administrator\.nuget\packages\microsoft.build.tasks.git\1.0.0-beta-62925-02\build\Microsoft.Build.Tasks.Git.targets 20

If you know something about it. Thanks in advance

timverwaal, 2020-07-07:
Seems to me this unrelated to the issue. Something with git and/or Nuget that causes the fault.

ZombieAllen, 2020-07-10:
> Seems to me this unrelated to the issue. Something with git and/or Nuget that causes the fault.

Thanks Mate

mycroes, 2020-09-04:
@ZombieAllen I ran into the same issue when updating the build infra, but that should be fixed. Also, @timverwaal's PR is merged, so I'm closing this issue. Feel free to reopen if you still run into issues.

closed 2020-09-04 by mycroes