S7NetPlus/s7netplus #271: C# ReadClass S7.net

VerbovenNick, 2020-07-06:
Hi

I am using the S7.net library in C# in order to read data from a VIPA PLC (similar to S7-300).  The C# project runs on a virtual machine (Windows 10 Pro, 64 bit) and is programmed in Visual Studio 2019.  I am reading a DB by the ReadClass method, all works fine for some hours, and then all of a sudden the ReadClass method returns an exception.
I could take a snapshot of 2 exceptions that i have seen (see attachment), I don't know which one was thrown first.
Has anybody experionce with this problem.
Note: I did not stop the S7oiehsx64 service, is that necessary ?

[image]
[image]

Thanks

FalcoGoodbody, 2020-07-09:
I have a VIPA PLC as well and I do have a similiar problem but I know the reason.
Here is another application which is reading 2 values via an OPC-Server from this PLC. The OPC-Server reads the data in a very short cycle (<200ms) and it seems as if this blocks all other communication with the PLC. It is even not possible to connect with the PLC via the PLC-IDE from Siemens (Simatic Manager).

Can you check if another application is reading data from your PLC?

VerbovenNick, 2020-07-10:
Hi

Thank for the answer.  My problem was caused accidently by triggering the console, after that some methods gave an exception.  Now I have a stable communication for about 3 days without any exception.

Kind regards

FalcoGoodbody, 2020-07-21:
Can you please explain this a little bit more detailed? It would be nice to know for me.

Thank you very much.

VerbovenNick, 2020-07-27:
Hi

 

I had some plc ReadClass exceptions when I triggerd the C# Console.  When triggering the console I assume that the application is not running, after “undoing” that console trigger by ESC, I had some exceptions.  I will not investigate this item further because the console will never be triggerd.

 

The application is running now for 3 weeks in production and everything seems to go good.  My application collects plc data with the readclass method, and the sends data to SQL or webservice in order to complete  traceability on the production line.  Once I had plc readclass exceptions because the touchpanel linked with the vipa plc was restarted again.  My reconnect procedure worked.   I think this is a exceptional situation, so I will not investigate this further.

 

If you have any questions, please contact me.

 

Nick

mycroes, 2020-09-08:
Closing this because the issue seems fixed. Also, S7NetPlus 0.7.0 has improved network communication and better exception messages on failures, so please upgrade first if issues persist.

closed 2020-09-08 by mycroes